realsenseai/librealsense #9451: Pyrealsense2 can not connect to the camera on WSL2

YuZiGu, 2021-07-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 running **WSL2** Linux Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.4.72-microsoft-standard-WSL2  | 
| Platform                                 | PC |
| SDK Version                            |            | 
| Language                            |  python                 | 
| Segment			|  AR                      | 

### Issue Description

My WSL2 system can not connect to the camera while using pyrealsense2. 

```
  File "ar_allTags.py", line 23, in <module>
    pipeline.start(realsense_cfg)
RuntimeError: No device connected
```

Is there a way to connect the camera to a WSL2 system?

MartyG-RealSense, 2021-07-20:
Hi @YuZiGu  My understanding is that WSL2 does not have USB device support, and so I would therefore recommend installing the pyrealense2 wrapper on Windows if you do not have access to a Linux computer or do not have a dual-boot operating system setup for Windows and Linux on your computer.

https://docs.microsoft.com/en-us/windows/wsl/wsl2-faq#can-i-access-the-gpu-in-wsl-2--are-there-plans-to-increase-hardware-support-

https://github.com/IntelRealSense/librealsense/issues/8217#issuecomment-764800820

MartyG-RealSense, 2021-07-27:
Hi @YuZiGu  Do you require further assistance with this case, please?  Thanks!

YuZiGu, 2021-07-27:
Then, I guess it is now possible to work on with RealSense with WSL2. Thanks, @MartyG-RealSense!

closed 2021-07-27 by YuZiGu