realsenseai/librealsense #9452: Couldn't resolve requests - "self.pipeline.start(config)"

jamesmarlin, 2021-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|--------------------------------|------------------------------------------- |
| Camera Model                       | L515          | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                  | 
| Platform                                 | PC             |
| SDK Version                            |                  | 
| Language                               | python      | 
| Segment			        |                    | 

### Issue Description
Hello!
I'm currently trying to use "box_dimensioner_multicam__demo.py" with my LIDAR L515, but I continue to receive the same error: 
Traceback (most recent call last):
  File "C:\Users\james\Downloads\measure_object_distance\Boxtest.py", line 255, in <module>
    device_manager.enable_all_devices()
  File "C:\Users\james\Downloads\measure_object_distance\Boxtest.py", line 131, in enable_all_devices
    self.enable_device(serial, enable_ir_emitter)
  File "C:\Users\james\Downloads\measure_object_distance\Boxtest.py", line 117, in enable_device
    pipeline_profile = pipeline.start(self._config)
RuntimeError: Couldn't resolve requests

*I changed the name of the .py file to be called Boxtest*

From what I've read on other issues similar to mine in this repo, the solution should be changing the resolution (Ex: c.enable_stream(rs.stream.depth, 320, 240, rs.format.z16, 6) ), but this is not working for me. Is there something I am missing?

MartyG-RealSense, 2021-07-20:
Hi @jamesmarlin   The error **Couldn't resolve requests** indicates that the SDK was requested to provide a resolution / FPS mode that it could not deliver, either because the mode is not currently supported by that particular camera model or because the request was not understood (perhaps because of a typing error).

The list of resolutions supported by L515 at the time of writing this is shown in the image below.

[image]

For the benefit of RealSense users reading this case at a future date: firmware **1.5.8.1** or newer is required for support of RGB resolutions 640x360 and 640x480 on the L515 camera model.

In your particular case @jamesmarlin it is likely it is the FPS that is causing the **Couldn't resolve requests** message if you are using 320x240 depth at 6 FPS.  This is because the only FPS supported for depth on L515 is **30**.

[image]

jamesmarlin, 2021-07-20:
Hi @MartyG-RealSense ! Apologies for the late repsonse. Thank you for your response and adding those labels. I tried changing it 30fps, but no luck.

MartyG-RealSense, 2021-07-20:
It looks as though the scripts have been heavily modified and it is not just a script name-change.  Is that correct, please?  If it has been significantly modified then this is not an ideal project to modify, as previous attempts to integrate parts of it (especially realsense_device_manager.py) into RealSense users' own custom applications have often failed.  It has been successfully done but it is not easy.

jamesmarlin, 2021-07-21:
@MartyG-RealSense Hello again. Last night I thought I may have gotten a modified version in an attempt to find the correct version, but this may have not been the case. I may just put a pause on this version and try out rs-measure in the SDK. Or do you know of any other python counterparts of rs-measure?

MartyG-RealSense, 2021-07-21:
There is a Python conversion of rs-measure at the link below that a RealSense user created.  They later expanded the project into a much larger measuring project, though the original **measure_new.py** script linked to below is a standalone function that can be separated out.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

closed 2021-08-02 by MartyG-RealSense

MartyG-RealSense, 2022-10-11:
Hi @PierreLouisT  I believe that your script is defining multiple cfg stream configurations and commenting out some of them with the # symbol so that they are ignored.  Is that correct, please?

```
rs.config.enable_device_from_file(config, args.input) 

# config.enable_stream(rs.stream.accel) 
# config.enable_stream(rs.stream.gyro) config.enable_stream(rs.stream.depth, rs.format.z16, 15) 
# config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 15) 

config.enable_stream(rs.stream.color, rs.format.rgb8, 15) 
config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 15) 
config.enable_stream(rs.stream.infrared, 2, 1280, 720, rs.format.y8, 15) 

cfg = self.pipeline.start(config)
```

Looking at the uncommented cfg instructions, they would be likely to produce a **Couldn't resolve requests** error if the camera was detected as being on a USB 2.1 connection instead of USB 3.  The range of supported resolutions and FPS speeds on a USB 2.1 connection is more restricted on USB 3, so the only FPS speed supported on USB 2.1 for 1280x720 resolution would be **6** (5 on the D455 model) instead of 15. 

Also, Infrared 2 is not supported on a USB 2.1 connection.

PierreLouisT, 2022-10-11:
Sorry, Indeed I tried to put in commentary accel and gyro stream only because it doesnt work that way. 
I checked USB connection and it is 3.2 as well

PierreLouisT, 2022-10-11:
When I only use RGB stream, depth, and both IR it works. But how to configure IMU data ?

MartyG-RealSense, 2022-10-11:
As you are using a bag file as the data source, you should avoid requesting streams or a stream configuration that are not in the bag file.  The streams requested and their resolution / FPS should be the same as the configuration that the streams in the bag were recorded at.

Can you confirm please that the bag file contains both gyro and accel streams in its recording?

PierreLouisT, 2022-10-11:
Yes I confirm but I know why I got the mistake, I put the wrong resolution in the RGB stream, it can explain why I got this error,
Thank you for your help

MartyG-RealSense, 2022-10-11:
You are very welcome  :)

surajbalore, 2023-05-10:
Hello ,
I am getting following error for realsensed d455. I can see images for RGB but having problem for depth images.

 pipeline.start(config)
**RuntimeError: Couldn't resolve requests**

thank you.

surajbalore, 2023-08-06:
> *Hello MartyG-RealSense Thank you for fast response I found error it was
> typo mistake i was trying to get depth image using colour stream. Thank you
> so much for help.*
>
> You are very welcome, @surajbalore <https://github.com/surajbalore>

Hello,
I hope you are doing well.
I would like you to ask that if I want to create a model to compare the
images from realsense d455 with helios2 , which aspect or point should I
use?
I would love to take it into consideration and thank you in advance for
your time and help.

Regards,
Suraj Balore

MartyG-RealSense, 2023-08-06:
Hi @surajbalore  If you wanted to compare differences between two images then the free open-source **CloudCompare** tool might be useful once the images had been converted into point clouds

https://www.danielgm.net/cc/

https://www.cloudcompare.org/doc/wiki/index.php/Introduction

It might not be a useful like-for-like comparison though as the two cameras use different depth sensing technologies.  Helios2 is apparently a **time of flight** camera, whilst D455 is **stereo depth**.  The closest RealSense camera model in principle to Helios2 is the L515, which is a **lidar depth** camera.  The L515 model is now retired though.

surajbalore, 2023-08-29:
Hello Marty,
I hope you are doing well. I have some doubt regarding Helios2 as i am not
able to take pictures on laptop by using ArenaView as well as python
program as I am getting error for

import pyhelios

camera_hel = pyhelios.Camera()
camera_hel.open()

try:

    image = camera_hel.capture_image()

    file_path = 'image.jpg'
    image.save(file_path)
    print(f"Image saved to {file_path}")

finally:

    camera_hel.close()

Traceback (most recent call last):
  File "C:\Users\Suraj\PycharmProjects\helios\main.py", line 4, in <module>
    camera = pyhelios.Camera()
AttributeError: module 'pyhelios' has no attribute 'Camera'

and if I try using ArenaView it says no camera detected. I am kinda
confused because it's an ethernet cable and I am using an ethernet to Usb
connector.
Can you kindly tell me what's the problem?

MartyG-RealSense, 2023-08-29:
It is possible to access the RGB stream in Windows with a non-RealSense application, like the image from an L515 in the **Camera** Windows app below.

[image]

If you are trying to access the depth stream though with an application that is not RealSense compatible then it would not be displayed.

********

Another circumstance in which you would not be able to access a particular camera stream is if you already have another application running that is using that stream.  For example, if you accessed the RGB stream with one program and then run another program that also wanted to capture color from the camera, the second program would not be permitted to access the RGB stream as the first program has already claimed exclusive access to it and another program can only use that stream if the first program relinquishes its claim on it (for example, by stopping its color stream or the program closing down).

*********

As far as I am aware, RealSense cameras have not been tested on an ethernet-USB adaptor.It is more typical for the USB and ethernet cabling to be separate, like in the Python **EtherSense** RealSense ethernet networking project.

https://github.com/krejov100/EtherSense

[image]

surajbalore, 2023-12-07:
> Hi @surajbalore <https://github.com/surajbalore>, were you able to
> compare performance between the D455 and Helios2 ?? I am really interested
> in your results.

Hello Sir,
First of all so sorry for late reply .I am working on other assignment
right now and so that task is on hold but i would like to ask for your help
for some doubts i have which are follows.
1) I want to calculate weight of person using realsense , do you have any
data which can be useful for it.
2) I want to remove the noises from depth image,I am using bilateral and
spatial but i am not satisfied with it is there any other filtor.

If you have any answer or data regarding above points please let me know
thank you in advance.

Regards,
Suraj Balore

MartyG-RealSense, 2023-12-07:
Hi @surajbalore  You can measure the **volume** of something with a RealSense camera, but it would not be able to calculate weight.  For example, you could measure the volume of a mannequin dummy of an adult person but that dummy would weigh much less than a real human adult.

In regard to depth image noise, you could first try applying a Temporal filter and setting its **Smooth Alpha** option's value to '0.1'.  This can reduce fluctuation of depth values and so make the depth image less noisy.

surajbalore, 2024-01-24:
Hello ,
I was hoping to get some guidance from your side. I want to access helios 2
lider camera using python(its connected by ethernet).I only have mac
address and dont have ip address(I dont have admin access to download any
app and using command promt i tried but it was not helpful). Can you please
help me regarding it. If you have any experience with Helios 2 or have any
data regarding it please share with me.
Thank you.

regards,
Suraj

MartyG-RealSense, 2024-01-24:
Intel are not able to provide support for non-RealSense cameras, unfortunately.  Although we cannot assist in this case, it may be worth it for you to explore the 'Arena SDK' for Lucid's cameras (which include the Helios2 time of flight model).

https://thinklucid.com/arena-software-development-kit/