realsenseai/librealsense #9453: Can't calibrate D455 cameras.

MatthewLKomar, 2021-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Unity         | 
| Segment			|             Intel Real Sense Viewer               | 

### Issue Description
[image]
[image]
[image]

I can't calibrate my intel D455. I tried Intrinsic OCC, Extrinsic, OCC Extended. It either times out or doesn't start the calibration at all. 
I get this error: UNKOWN in rs2_run_on_chip_calibration_cpp Not enough depth pixels! (Fill_Factor_LOW). Please retry in different lighting conditions. 

I've tried this on different machines and in different environments of various lighting and I still get this error. 

Tare Calibration doesn't work on walls past 4 feet away. I can't Calculate Ground Truth. It always tells me to adjust the camera position and tells me to put the target in the middle of the camera image. 

I've attached the camera calibration settings.
[realsense_cameracalibration.txt](https://github.com/IntelRealSense/librealsense/files/6852227/realsense_cameracalibration.txt)

MartyG-RealSense, 2021-07-21:
Hi @MatthewLKomar  I tested your scenario with a D455 and was able to always achieve a **Good** calibration result when calibrating in "Slow" mode.  I was though able to replicate your error message if calibrating in "White Wall" mode.  Are you using White Wall, please?

[image]

If you are using White Wall, this mode was designed to aid on-chip calibration with the **D415** camera model and is not required for D435 (and by extension the D455, though the paper does not mention that model).  The paper says "On-chip calibration of D435 cameras does not require the “white wall” mode regardless of target type and projector setting; one of the standard speed settings is recommended under most conditions".

https://dev.intelrealsense.com/docs/self-calibration-for-depth-camera

MatthewLKomar, 2021-07-21:
Hello @MartyG-RealSense 

I think this is the link you're refering to: https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras
I've read the page several times over the past couple days, but I couldn't manage to do the C++ code. 

How can I adjust target settings? 
How can I turn on the projector? The documentation doesn't say how to turn it on. 

The documentation linked above provided this code, which doesn't work as the functions mensioend are missing parameters: 
[image]
Here are this missing parameters.
[image]
[image]
[image]
[image]
How can I get them? 

I'm not using White Wall calibration method. Is there some exact steps for your calibration, as I can't seem to calibrate all 3 of my cameras (all brand new) regardless of the computer and environment they're in. 

The slow calibration, as well as all the others, usually either time out (they keep running, past 100%) and eventually fail or they fail under "Not enough depth pixels!". 

I've also tried Factory Reseting them in the SDK real sense viewer, but it wouldn't change anything. Should I format the cameras in C++?

MartyG-RealSense, 2021-07-21:
Brand new cameras should not yet need calibrating, as they are calibrated in the factory during the manufacturing process and so usually have very good calibration when they are new out of the box after purchase.

Control of the camera's internal projector is not part of the calibration system, but it can be controlled with C++ code.  The SDK documentation page at the link below provides example code under the 'librealsense2' heading for controlling the on-off status of the projector and the **Laser Power** level.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

A factory reset would return the calibration to its factory-new state and so would not be very different if the cameras are brand new anyway.

A RealSense user shared their C++ script for on-chip calibration control in the link below.

https://github.com/IntelRealSense/librealsense/issues/8337#issuecomment-777040007

MatthewLKomar, 2021-07-21:
Hi Marty, 

I managed to calibrate all the other cameras, except the one with which I had a problem initially (before calibration started). Factory resetting & calibration didn't return it to usable state so I believe it's defective. 

Thanks a lot for your help.

closed 2021-07-21 by MatthewLKomar

MartyG-RealSense, 2021-07-22:
You are very welcome @MatthewLKomar - it's excellent to hear that you had success with calibration.  Thanks for the update!