realsenseai/librealsense #9454: multi usb cameras (d435i and others)

jjuhee, 2021-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.04.54  | 
| Operating System & Version |   Ubuntu 16.04.4 LTS  | 
| Kernel Version (Linux Only)    | 4.9                                        | 
| Platform                                 | sda845  |
| SDK Version                            |  2.29.0                       | 
| Language                            |  c++                          | 
| Segment			|  Robot                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>

HW condition : 
I'm using two d435i and three usb cameras(logitech c930e).
hw connection is like below.
( d435i 2ea ) -> 3.0 usb port -> pcie   ───────> sda845
( c930e 3ea ) -> 3.0 usb hub -> 2.0 port ─┘

I enabled one d435i RGB stream (848*480, 30fps). (didn't enable the other d435i yet.) 
c930e 1,2,3 are streaming RGB (mjpg 960*540, 30fps). 
 
I am capturing every 66ms,
Enabling one depth stream of two d435i, sometimes causes frames to come too late. (over 120ms.)
With all depth stream enabled, frames come out very late.!
Is there any way to improve it?
I want to enable at least one depth stream of the two d435i. 
and i hope all frames come out in time. (66ms)

MartyG-RealSense, 2021-07-21:
Hi @jjuhee  You stated that you enabled the RGB stream of a D435i at 848x480 and 30 FPS.  Then you enable 'one depth stream of two D435i'.

Are you using both RGB and depth on one of the D435i and finding that the depth FPS rate drops to half, please?

jjuhee, 2021-07-21:
i'm sorry for not explaining properly. 
if i enabled both RGB and Depth of one d435i, then c930e device is often slow to receive frame. 

i'm using 4 threads for each camera device, and measured the time to receive the frame.
(forget about the other d435i. (connected but not using), so one d435i , three c930e) 

i mesured the time between 1 and 2 marked below. :(
 
getRSFrameThread()  { // thread for d435i
1
*sensor_ok = pipe.try_wait_for_frames(&fs, 1000U);
2
}

getMultiUVCFrameThread()  { // thread for c930e
1
bRet = m_vstUVCVideoCaptures[idx].oVideoCapture.grab();  // opencv VideoCapture API
2
}

<measured time>

- good case :

[03-01 02:45:13.831692] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 1 Check time duration(us) : **291** / 66000
[03-01 02:45:13.831893] <fatal> [RS2] getRSFrameThread() Check time duration(us) : **36** / 66000
[03-01 02:45:13.850964] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 2 Check time duration(us) : **894** / 66000
[03-01 02:45:13.867213] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 0 Check time duration(us) : **534** / 66000

[03-01 02:45:13.886333] <fatal> [RS2::UVC] Check time duration(us) : **56398** / 66000 // total duration measured in main function.

- bad case :

[03-01 02:45:19.971425] <fatal> [RS2] getRSFrameThread() Check time duration(us) : **34** / 66000
[03-01 02:45:19.971894] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 0 Check time duration(us) : **638** / 66000
[03-01 02:45:19.989775] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 1 Check time duration(us) : **722** / 66000
[03-01 02:45:20.127627] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 2 Check time duration(us) : **119637** / 66000

[03-01 02:45:20.153968] <fatal> [RS2::UVC] Check time duration(us) : **183027** / 66000

MartyG-RealSense, 2021-07-21:
Thanks very much for the clarification and additional detail.  I researched the **c930e** and see that it has a lighting auto-correction feature called **RightLight**.  It may be worth turning the D435i away from the c930e if they are facing each other to see whether that improves the c930e's timing.  This is because the D435i's **IR emitter** component might be causing disruption if the laser is projecting light into the c930e 's field of view.

jjuhee, 2021-07-21:
they are not facing each other.

MartyG-RealSense, 2021-07-21:
Are the D435i and c930e cameras synced in any way or do they opoerate completely independently?

jjuhee, 2021-07-21:
how can i know that they synced ?
i just wait for all threads to join() and use.

MartyG-RealSense, 2021-07-21:
If you are not deliberately syncing the cameras then they would not be synced and would be operating independently.

For example, the link below has a past case of a RealSense user who deliberately synced their RealSense camera with a non-RealSense camera using **camera metadata**.

https://github.com/IntelRealSense/librealsense/issues/2186

jjuhee, 2021-07-22:
I don't know if I can check it this way. 
I checked CAP_PROP_POS_MSEC in opencv api. 
In the good case, the time difference between each thread is about 50ms, 
but in the bad case, it seems to increase to 200~300ms.

- good case:

[03-01 00:02:28.811777] <fatal> [RS2] getRSFrameThread() Check time duration(us) : 33 / 66000
[03-01 00:02:28.812196] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 0 Check time duration(us) : 505 / 66000
[03-01 00:02:28.830648] <info> [RS2::UVC][jhee] idx 0 CAP_PROP_POS_MSEC: **143571**
[03-01 00:02:28.832606] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 2 Check time duration(us) : 349 / 66000
[03-01 00:02:28.846266] <info> [RS2::UVC][jhee] idx 2 CAP_PROP_POS_MSEC: **143514**
[03-01 00:02:28.847819] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 1 Check time duration(us) : 454 / 66000
[03-01 00:02:28.863777] <info> [RS2::UVC][jhee] idx 1 CAP_PROP_POS_MSEC: **143648** 
[03-01 00:02:28.870354] <fatal> [RS2::UVC] Check time duration(us) : 58884 / 66000

- bad case:

[03-01 00:02:30.549906] <fatal> [RS2] getRSFrameThread() Check time duration(us) : 41 / 66000
[03-01 00:02:30.748760] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 1 Check time duration(us) : 198820 / 66000
[03-01 00:02:30.768686] <info> [RS2::UVC][jhee] idx 1 CAP_PROP_POS_MSEC: **145641** 
[03-01 00:02:30.771193] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 0 Check time duration(us) : 705 / 66000
[03-01 00:02:30.788643] <info> [RS2::UVC][jhee] idx 0 CAP_PROP_POS_MSEC: **145101** 
[03-01 00:02:30.790122] <fatal> [RS2::UVC] getMultiUVCFrameThread() idx 2 Check time duration(us) : 335 / 66000
[03-01 00:02:30.803618] <info> [RS2::UVC][jhee] idx 2 CAP_PROP_POS_MSEC: **145439** 
[03-01 00:02:30.807423] <fatal> [RS2::UVC] Check time duration(us) : 258636 / 66000

MartyG-RealSense, 2021-07-22:
Referring back to your original diagram about the USB configuration:

[image]

I note that the c930e cameras are connected to a USB 3.0 hub but the hub is apparently connected to a USB **2.0** port.  That is likely to result in a speed bottleneck, as the hub would be treated as a slower USB 2.0 device.  It is like how a device on a USB 3 cable that is plugged into a USB 2 port will be treated as USB 2.  A USB 3 device needs to be connected to a USB 3 port to take advantage of its full capabilities.

jjuhee, 2021-07-26:
c930e is actually USB2.0 supported device. So I use 2.0 hub also.

jjuhee, 2021-07-26:
could it be a bandwidth issue ?
Is it possible to get RGB camera data in Y8 format from C3D to reduce the amount of data?

MartyG-RealSense, 2021-07-26:
The RealSense Viewer's **Add Source** button allows the adding of sources that are not RealSense camera models, such as the built-in webcam of a laptop (see the image below).  I do not have an external USB webcam to test with though.

[image]

MartyG-RealSense, 2021-08-01:
Hi @jjuhee  Do you require further assistance with ths case, please?  Thanks!

jjuhee, 2021-08-03:
Sorry to late. 
but I don't know your intentions.
I can attach the webcams with the latest version of Realsense viewer. what should I do after that ?

MartyG-RealSense, 2021-08-03:
I do not have an ordinary webcam to test with.  But I would assume that if the RealSense Viewer is able to detect the webcams then they would be listed if you click on the **Add Source** button at the top of the Viewer's options side-panel.  You could then left-click on a webcam in the list to add it to the Viewer.

closed 2021-08-15 by MartyG-RealSense