realsenseai/librealsense #9455: L515 not recognized when boot up PC

Consultaing, 2021-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.4.1| 
| Operating System & Version | Linux (Ubuntu 18.04.3 LTS)| 
| Kernel Version (Linux Only)    | 4.15.0-66-generic x86_64                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.48.0-0~realsense0.4975                          | 
| Language                            | python                         | 
| Segment			| Robot | 

### Issue Description

Hello to the Intel Realsense technical team,

We have 4 intel realsense cameras for different purposes in our project.
There are 2 D415 cameras, 1 D435 camera, 1 L515 camera.

The D4xx cameras all work properly when the PC is turned on or rebooted. They are detected both by the operating system between the devices, as well as by the library.

However, in the LIDAR L515 camera we have detected that when the PC is turned on or restarted with the camera plugged into the USB port, it is not recognized by the operating system. (It does not appear in the list of devices when we use the lsusb or usb-devices command)

When we use the library, it shows us the following error.

	pipeline_profile = config.resolve (pipeline_wrapper)
	RuntimeError: No device connected

If we unplug the L515 camera and plug it back in, it is detected among the USB devices and works correctly without any error.

This topic has been previously covered in the D4xx series cameras as can be seen in the following issues:
#4457
#5913

On the other hand, we found a similar problem in another issue #4552, where they proposed a workaround to disable USB ports by software. This solution did not worked for us because to be able to reset it is necessary to detect the device (This topic is discussed in the issue).

For our project, this situation is **blocking** since the USB port is not easily accessible. The only solution on the user side is to have a USB card with independent power management for each port to be able to perform a software disconnection.

We have done tests on 3 different machines, 2 quite new machines but without independent port control, and 1 machine with independent port control.

**Please we need that firmware of the L515 camera to be checked to fix this error.**

I hope that description of the problem is clear enough to help solve it.

Thanks in advance for your kind help.

MartyG-RealSense, 2021-07-21:
Hi @Consultaing  As a starting point in investigating your case, could you try unplugging the USB cable at the camera and of it, turning the connector in the opposite direction and re-inserting it (the cable has two-way insertion at the camera end).  If you then shut down the computer and restart it with the power button, is the L515 now recognized when boot has completed?

Consultaing, 2021-07-21:
Hello @MartyG-RealSense,

Now we have tested your suggestion to change the position of the USB type C connector. We have tried both positions on two machines and the error persists.

If you need any more test from our side, do not hesitate to tell us.

Thank you.

MartyG-RealSense, 2021-07-21:
Is the L515 camera detectable at boot if it is the only camera attached and the D4xx cameras are all unplugged?

Consultaing, 2021-07-22:
Initially we had all the cameras connected when we discovered the error, but all the tests that I commented in this issue were done only connecting the L515.

The L515 camera connected with the keyboard and the mouse is not detected when turn on the PC.

Consultaing, 2021-07-22:
Hello @MartyG-RealSense,

We have found a solution to our problem, although we believe that it really is a workaround.

We have disabled automatic suspension of USB ports when devices are idle. In this way we turn on the PC with the USB ports always active.

We have modified in Ubuntu 18.04 LTS the Grub flag **usbcore.autosuspend** by default is 2 (2000 ms) to **-1** which disables the automatic suspension.

```
sed -i 's/GRUB_CMDLINE_LINUX_DEFAULT="/&usbcore.autosuspend=-1 /' /etc/default/grub

update-grub

systemctl reboot
```

We have turned on and restarted the PCs several times, and the L515 camera has worked correctly.

On the other hand, the solution that we discussed at the beginning of using a USB card with independent management of the power of the ports would also be valid with the automatic suspension activated.

In any case, we think that the technical team at intel realsense should review this case to improve the behavior of the L515 camera.

Thank you for your help!

MartyG-RealSense, 2021-07-22:
You are very welcome!  It's excellent to hear that you were able to develop solutions.  Thanks very much for sharing your methods, which I will pass to Intel.

MartyG-RealSense, 2021-07-28:
Case closed due to solution achieved and passed to Intel and no further comments received.

closed 2021-07-28 by MartyG-RealSense