realsenseai/librealsense #9457: How can I get depth_frame?

hyebin-rdv, 2021-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | ubuntu 18.04  | 
| Kernel Version (Linux Only)    |5.4.0-77-generic| 
| Platform                                 | PC  |
| SDK Version                            | 2.45.0 | 
| Language                            c++ | 

### Issue Description
This is my code.
`    using namespace cv;

	auto frames = m_pipe.wait_for_frames();
    auto aligned_frames = m_align->process(frames);

    aligned_depth_frame = aligned_frames.get_depth_frame();
    color_frame = aligned_frames.get_color_frame();

    const int w = aligned_depth_frame.as<rs2::video_frame>().get_width();
    const int h = aligned_depth_frame.as<rs2::video_frame>().get_height();
    
    cv::Mat depth_mat = frame_to_mat(aligned_depth_frame);
    
    return depth_mat ;    `

I got an image, but it's not the image I want. 
This is the image I want.
[image]

However, the output is as follows.
[image]

How to get depth image I want?

hyebin-rdv, 2021-07-21:
First image is cpature iamge to use realsense-viewer

MartyG-RealSense, 2021-07-21:
Hi @hyebin-rdv  If you are aiming to recreate the colorized depth image from the Viewer then you should not need to use alignment to accomplish that.  The SDK's OpenCV C++ example program **rs-imshow** can visualize colorized depth depth by applying a **colorizer**.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/imshow

hyebin-rdv, 2021-07-21:
Thank U!!!!
one more question.
[image]

I don't know what to put in there.
BGR8? BGR16?

MartyG-RealSense, 2021-07-21:
If it is an RGB color image and ROS is being used, i would recommend trying **RGB8**.  Otherwise, the depth format is always **Z16**

hyebin-rdv, 2021-07-21:
Nothing comes up......
[image]

hyebin-rdv, 2021-07-21:
[image]
--> publisher
[image]
 Neither rgb8 nor bgr8 floats.

MartyG-RealSense, 2021-07-21:
This subject of OpenCV coding in ROS is not a subject that I have expert knowledge of unfortunately.  Comparing your script to one in the link below though, I can't see anything obvious that you are doing wrong.

https://stackoverflow.com/a/34618037

hyebin-rdv, 2021-07-21:
Ok, I got it. Thanks to help me!

closed 2021-07-21 by hyebin-rdv