realsenseai/librealsense #9459: Black holes and invalid link to keras model (for hole filling)

Basti110, 2021-07-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | l515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 20  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  python                        | 
| Segment			|  others                          | 

### Issue Description
Hello, 

today I received the realsense and recorded some depth images. I want to train a machine learning model with the depth information.  Therefore I need images like in this data set [Nyu](https://cs.nyu.edu/~silberman/datasets/nyu_depth_v2.html). My first problem are many invalid pixels on black objects (like displays, tables, chairs). Is there a better configuration for the realsense to receive better results? 

For the denoising and hole filling, I want to use the Keras model like described in the examples. But I can not download the [Keras Unet model](http://realsense-hw-public.s3.eu-west-1.amazonaws.com/rs-tests/ML/Depth_Learning/DEPTH_Keras_Unet.model.zip) from git [examples](https://github.com/nohayassin/librealsense/tree/tensorflow/wrappers/tensorflow) Part 5. All links are invalid on this page (AccessDenied). Can somone provide a valid link to this model?

MartyG-RealSense, 2021-07-21:
Hi @Basti110   It is a general physics principle (not specific to RealSense) that dark grey and black objects can be difficult for the camera to read depth information from because such colors absorb light.  The darker the shade, the less light that is reflected back.  Projecting a strong light source onto such objects can help to bring out depth detail, as demonstrated in the L515 case linked to below where areas that light reflects from are rendered in blue instead of empty black areas of no depth information.

https://github.com/IntelRealSense/librealsense/issues/6757

Holes may also be reduced when using L515 if there is a wall behind the object to bounce light off.

In regard to the broken link, I will tag the provider of the model file into this discussion.  Hi @nohayassin  Can you provide a working link for the **DEPTH_Keras_Unet.zip** model file for Example 5 of the TensorFlow wrapper please?  Thanks!

Basti110, 2021-07-22:
Okay Thank you :) 
I will try to play with the light sources a bit. Maybe together with the denoising, the results are usable for me.

nohayassin, 2021-07-23:
Hi @Basti110 , the url had changed to https://librealsense.intel.com/rs-tests/ML/Depth_Learning/DEPTH_Keras_Unet.model.zip
I will fix it, thanks for your notice.

closed 2021-08-03 by MartyG-RealSense