realsenseai/librealsense #9460: Displaying text with rs-measure

jamesmarlin, 2021-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | LIDAR L515| 
| Firmware Version                   | 01.05.08.01|
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC |
| SDK Version                            |                           | 
| Language                            |  Python                        | 
| Segment			     |                       | 

### Issue Description
Hello! I'm trying to use rs-measure, but the box that is displayed when changing measurements does show any numbers. I assume it's supposed to because it changes in length as I measure things with my camera. I feel like I'm glossing over what needs to be changed to make this work, so what am I missing here?

MartyG-RealSense, 2021-07-21:
Hi @jamesmarlin   I tested **rs-measure** in Windows with D415, D435i, D455 and L515 and confirmed that values were not displaying for the defined line, just a grey rectangle.  

[image]

The built-in **Measure** tool in the 3D point cloud mode of the **RealSense Viewer** tool displayed the measurement correctly though.

[image]

jamesmarlin, 2021-07-21:
@MartyG-RealSense  Oh, I see! I thought it was just me! Thank you!

MartyG-RealSense, 2021-07-22:
Hi @jamesmarlin  After discussion with Intel, an official bug report about this issue with rs-measure has been filed with Intel.  This case should be kept open whilst the bug report is active, but you do not need to take any further action.  Thanks again for the report!

MartyG-RealSense, 2021-11-01:
Hi @jamesmarlin  Intel's bug report has been concluded.  The reported bug should be resolved in the forthcoming **2.50.0** RealSense SDK version.  I will now close this issue.  Thanks for the report!

closed 2021-11-01 by MartyG-RealSense