realsenseai/librealsense #9461: example.hpp can't open (beginner)

FlockaFlam3, 2021-07-21:
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.48.0 | 
| Language                               | C++ | 
Visual Studio 2017
### Issue Description
I am an absolute beginner to libraries and vcpkg. I've installed vcpkg and librealsense2 and then when I follow sample code for [#include "example.hpp"], I cannot open source file. I see that it is not in librealsense2, but in librealsesnse. Now I am wondering what to do. Do I install librealsense and link it somehow or is there some type of guide I can follow to show the sample code walk through and set up.

I have looked into this type of error, but nothing great has come out of that already.

MartyG-RealSense, 2021-07-21:
Hi @FlockaFlam3  If you have the option to do so, I would recommend installing the RealSense SDK on Windows by downloading and running the quick and easy automated installer file in the **Assets** file list of the SDK 'Releases' page.  The vcpkg installation method can be problematic.

The Assets list can be found at the bottom of the information listing for each SDK version.

https://github.com/IntelRealSense/librealsense/releases/

[image]

If you would prefer to use vcpkg then I will be happy to guide you through the process.

FlockaFlam3, 2021-07-21:
So after I've downloaded and installed, how do I get use of the sdk file in program files? And should I remove vcpkg?

FlockaFlam3, 2021-07-21:
Or is there a guide I can just follow from scratch to fully set up the coding stuff?

MartyG-RealSense, 2021-07-21:
After installation is complete, the installer will have placed launch shortcut icons for the **RealSense Viewer** and the **Depth Quality Tool** programs on your Windows desktop.  The RealSense Viewer especially is great for beginners as it is simple to use and provides a large range of camera control features in a single software package.

If you would like to explore SDK example programs, the installer also creates a folder of pre-built and instantly launchale versions of examples.  You can find these easily by right-clicking on the RealSense Viewer launch shortcut icon and selecting the **Open file location** menu option.  This will open the SDK's **Tools** folder, where you can browse through examples and launch them with a double-leftclick.

Some examples need to be opened in the Windows Command Prompt text interface instead of with a double-leftclick.  A good example program to start with in that folder that is launchable with a double-leftclick is **rs-capture**.  

Once you feel ready to progress to coding, a RealSense user has created a guide (in the link below) to setting up a Visual Studio project with the three **props** files provided in the root directory of the SDK folder.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

[image]

FlockaFlam3, 2021-07-21:
Thank you very much for all of the help!

closed 2021-07-21 by FlockaFlam3