realsenseai/librealsense #9463: kinfu重建效果不佳

Tony-le-han, 2021-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<48>.<0> }                          | 
| Language                            |  {C/C#/opencv/pcl                          | 
| Segment			|  {Kinfu }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
kinfu 重建时经常重建失败重置场景，并且点云也十分粗糙，有人知道解决办法吗？

MartyG-RealSense, 2021-07-22:
Translation: kinfu reconstruction effect is not good.  When kinfu rebuilds, it often fails to rebuild and reset the scene, and the point cloud is also very rough.  Does anyone know a solution?

*************

Hello @Tony-le-han    The quality of rs-kinfu point clouds is discussed in the link below.  Use of the High Accuracy camera configuration preset was suggested as a way to improve results.

你好@Tony-le-han 下面的鏈接討論了 rs-kinfu 點雲的質量。 建議使用 HighAccuracy 相機配置預設作為改善結果的一種方式。

https://github.com/IntelRealSense/librealsense/issues/3406

Tony-le-han, 2021-07-23:
> 翻译：kinfu重建效果不好。kinfu重建的时候，经常重建和重置场景失败，点云也很粗糙。有谁知道解决方案？
> 
> 你好@Tony-le-han 下面的链接讨论了 rs-kinfu 点云的质量。建议使用高精度相机配置预设和更改激光功率设置作为改善结果的方法。
> 
> 你好@Tony-le-han下面的链接讨论了 rs-kinfu 点云的质量。 建议使用高精度相机配置预设和更改激光功率设置作为改进结果的方法。
> 
> #3406
高精度相机配置应该是初始已经预设了吧， dpt.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_DENSITY);不知道是不是这样
还有一个问题：intel realsense viewer 中设置最大深度，最小深度，为什么会没有效果

MartyG-RealSense, 2021-07-23:
"The high-precision camera configuration should be preset at the beginning, dpt.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_DENSITY);   I don’t know if this is the case.  There is another question: why there is no effect if the maximum depth and minimum depth are set in the Intel Realsense Viewer"

There have been a couple of past cases in which RealSense users had an error with the HIGH_DENSITY preset part of the rs-kinfu script.  If you have an error then you can comment out this part of the kinfu script or try editing an OpenCV source file, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/8931#issuecomment-830063845

In regard to setting a maximum and minimum depth, I would recommend doing this with the Threshold Filter in the Post-Processing section of the realsense-viewer options panel if you are not using this method already.

**********

在過去的幾個案例中，實感用戶在 rs-kinfu 腳本的 HIGH_DENSITY 預設部分出錯。 如果您有錯誤，那麼您可以註釋掉 kinfu 腳本的這一部分或嘗試編輯 OpenCV 源文件，如下面的鏈接所述。

https://github.com/IntelRealSense/librealsense/issues/8931#issuecomment-830063845

關於設置最大和最小深度，如果您尚未使用此方法，我建議您使用 realsense-viewer 選項面板的後處理部分中的閾值過濾器來執行此操作。

MartyG-RealSense, 2021-07-29:
Hello @Tony-le-han  Do you require further assistance with this case, please?  Thanks!

你好@Tony-le-han 請問這個案例需要進一步的幫助嗎？ 謝謝！

closed 2021-08-03 by MartyG-RealSense