realsenseai/librealsense #9464: restore depth_frame to use .png and .csv file

hyebin-rdv, 2021-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.05.08.01 | 
| Operating System & Version | ubuntu 18.04 | 
| Kernel Version (Linux Only)    |5.4.0-77-generic | 
| Platform                                 |PC |
| SDK Version                            | PC| 
| Language                            | PC | 

### Issue Description
Hi, I got .png and csv to use "https://dev.intelrealsense.com/docs/rs-save-to-disk#section-expected-output"
[image]

I want to make a depth_frame video using these two. Use the appropriate depth_frame to extract and use point cloud later. 
Is there a way to restore the frame form using png and csv?

hyebin-rdv, 2021-07-22:
[image]
I want to make it into a form that can fit in here.

MartyG-RealSense, 2021-07-22:
Hi @hyebin-rdv  I would recommend avoiding loading in a PNG image file of a depth frame if possible.  Most of the depth information is lost when saving to PNG, and may involve the process of creating a frame from a synthetic source with the SDK's **software_device()** function (which is okay for C++ but has few references in Python for doing so).

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device
https://github.com/IntelRealSense/librealsense/issues/1274#issuecomment-370228022

Instead of loading a depth PNG, you could load a **.raw** file instead, as this image file format preserves depth information.A .raw is one of the three file formats (PNG, raw, csv) exported from the RealSense Viewer when using the depth stream's **Snapshot** frame saving option.

https://github.com/IntelRealSense/librealsense/issues/2534#issuecomment-430912589

I cannot think of a way to turn a csv file back into a depth frame though.

hyebin-rdv, 2021-07-23:
omg,,, I got it. Thanks.
Then, is there a way to restore it to the frame format("aligned_depth_frame"of the picture above) I want from raw file?

MartyG-RealSense, 2021-07-23:
The link below provides advice from a RealSense team member about aligning a depth .raw and a color .png by using software_device()

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412810987

closed 2021-08-03 by MartyG-RealSense