realsenseai/librealsense #9471: Depth_image cannot be refreshed properly

Larry0607, 2021-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D4345  | 
| Firmware Version                   | 05.12.14.50| 
| Operating System & Version |   Win10| 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            | 2.4.8                      | 
| Language                            |  python                          | 
| Segment			|  others                           | 

### Issue Description
I had some problems running the code in librealsense/wrappers/python/examples/opencv_viewer_example.py

### Question

1、The depth_image does not refresh when its FPS is 30.
`config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
`
2、The program reported an error when the depth_image FPS was 10.**

_Traceback (most recent call last):
  File "F:/Larry/Pro/RealSenseD435/Learn/2opencv_viewer.py", line 46, in <module>
    pipeline.start(config) # 以config的配置开始采集数据
RuntimeError: Couldn't resolve requests_

### The source code is as follows
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

pipeline.start(config)

try:
    while True:

        
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

       
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        depth_colormap_dim = depth_colormap.shape
        color_colormap_dim = color_image.shape

        if depth_colormap_dim != color_colormap_dim:
            resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
            images = np.hstack((resized_color_image, depth_colormap))
        else:
            images = np.hstack((color_image, depth_colormap))

        
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

   
    pipeline.stop()
```

MartyG-RealSense, 2021-07-23:
Hi @Larry0607  I cannot see an obvious reason in the script why depth would stop updating at 30 FPS, as it is just the same script as the original **opencv_viewer.example.py** without the comment lines.

In regard to the **Couldn't resolve requests** error: these occur because a requested stream configuration either was not currently supported at the time of running the program or there was a typing error that made the line unable to be understood by the SDK.  In this partcular case I would recommend using 6 or 15 FPS at 648x480 instead of 10, as 10 is not supported fo that resolution.

Larry0607, 2021-07-28:
> Hi @Larry0607 I cannot see an obvious reason in the script why depth would stop updating at 30 FPS, as it is just the same script as the original **opencv_viewer.example.py** without the comment lines.
> 
> In regard to the **Couldn't resolve requests** error: these occur because a requested stream configuration either was not currently supported at the time of running the program or there was a typing error that made the line unable to be understood by the SDK. In this partcular case I would recommend using 6 or 15 FPS at 648x480 instead of 10, as 10 is not supported for that resolution.

Thank you for your answer！The depth_image can be refreshed when I set 15 FPS at 640×480.However, the depth_image is still static when I set 30 FPS at 640×480.

MartyG-RealSense, 2021-07-28:
What is your Windows PC's CPU usage percentage like when depth is set to 30 FPS?  You can find out the usage by using the **Ctrl-Alt-Delete** keyboard shortcut to access the **Task Manager** interface and left-clicking on the **Performance** category.  Then view the CPU usage in real-time whilst your Python program is running.

[image]

closed 2021-08-09 by MartyG-RealSense