realsenseai/librealsense #9472: D435i深度测量精度

meiy-px, 2021-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot                   | 

### Issue Description
规格书上只提供了2m以内的深度测量精度，2m~10m的精度大概是多少？误差分布是否有规律？如都偏大或都偏小？

MartyG-RealSense, 2021-07-23:
Translation: D435i depth measurement accuracy.  The specification only provides depth measurement accuracy within 2m.  What is the accuracy of 2m~10m?  Is the error distribution regular? 

***************

Hello @meiy-px  A chart that estimates the **RMS error** in mm that increases linearly as distance from the camera increases is shown in the image below, with the D435 represented as the upper orange line.  This diagram is from the section of information about RMS error in Intel's camera tuning guide that is linked to below.

你好@meiy-px 下圖顯示了以毫米為單位估計 **RMS 誤差**的圖表，該誤差隨著與相機距離的增加而線性增加，D435 表示為上面的橙色線。 此圖來自以下鏈接的英特爾相機調整指南中有關 RMS 錯誤的信息部分。

[image]

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

meiy-px, 2021-07-27:
OK, Thx.

MartyG-RealSense, 2021-08-01:
Hi @meiy-px  Do you require further assistance with this case, please?  Thanks!

meiy-px, 2021-08-05:
@MartyG-RealSense We find that almost all D435i modules' test depth are bigger then groud truth. Is this normal?

MartyG-RealSense, 2021-08-05:
Hi @meiy-px  Yes, with the RealSense models in casings (D415, D435, D435i, D455) the depth is measured from the front glass, whilst on the caseless Depth Module Kit board versions it is measured from the lens.

The ground truth of RealSense depth measurements can be obtained by adding a minus-value to the measurements.  Pages 72 and 73 of the current edition of the data sheet document for the 400 Series cameras list the minus value to add for particular RealSense models.  There are two separate tables for the models with front glass and the caseless Depth Module versions.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

On the D435 and D435i models, **-4.2** mm should be added to measurements to find their ground truth value.

meiy-px, 2021-08-05:
Hi @MartyG-RealSense We have already minused 4.7mm. The test depth is still larger than ground truth. The chart below is the test result of one module, and the other modules also meet this tendency chart.

[image]

MartyG-RealSense, 2021-08-05:
If the red line is based on a real-world distance measurement (not a measurement with the camera's depth sensing) then it should be taken into account that there will be a certain amount of error factor in the camera reading due to a phenomenon called **RMS Error**.  This is where the amount of error in a depth measurement by the camera starts at aroud zero at the camera lense and increases linearly over distance.  

On the D435i model the drift in accuracy will become noticable at around 3 meters distance and beyond, as reflected in your own chart where the two lines start to diverge at the 3.000 point.

The D455 camera model has 2x the depth measurement accuracy over distance of the D435 / D435i, meaning that at 6 meters distance the D455 has the same depth accuracy as the D435 / D435i does at 3 meters.

You can find more information about RMS Error in point 5 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

meiy-px, 2021-08-06:
Yes, we know there is RMS error. We just don't know if the depth sensing result is likely to be smaller than ground truth. Or it is certainly larger than ground truth?

MartyG-RealSense, 2021-08-06:
The section of Intel's white-paper document about On-Chip calibration linked to below describes depth accuracy and measured distance versus ground truth distance.  The error factor can be + or - (under or over).

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-3-depth-accuracy

meiy-px, 2021-08-09:
Have you calibrated the camera before it leaves factory? Is the mothod the same as the self-calibration? What the distance do you choose to calibrate the camera?

MartyG-RealSense, 2021-08-09:
RealSense cameras are calibrated at the factory during the manufacturing process, yes.  I do not have information about the process used in the factory, though it is probably similar to the **OEM Calibration Target** board and its **OEM Dynamic Calibration** software that is sold in the official RealSense Store for $1500 USD that is aimed at use by manufacturing facilities and engineering departments.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

meiy-px, 2021-08-09:
OK, thx. After factory calibration, will you check the depth quality of the camera?

MartyG-RealSense, 2021-08-09:
I do not have information about the specific checking processes in the factory.  They are likely to be confidential anyway.

meiy-px, 2021-08-09:
The reason I asked the question is that some cameras are good within 2m. But there are obvious mistakes at long distance (5m or longer).

MartyG-RealSense, 2021-08-09:
That is the expected behaviour of the camera.  With RealSense 400 Series cameras, the **RMS error** - mentioned earlier in this discussion in https://github.com/IntelRealSense/librealsense/issues/9472#issuecomment-885414483 -  starts at around zero at the camera lenses and increases linearly as distance from the camera increases.  Drift in accuracy caused by RMS error becomes noticable at 3 meters and beyond.

meiy-px, 2021-08-09:
The mistakes include not only the depth accuracy, but also the angle of the plane and the depth uniformity of the left and right side of the plane. I can show you a test picture later.

MartyG-RealSense, 2021-08-09:
If your project is using the **get_distance** instruction to obtain depth then the effect of accuracy drift at the outer edges of an image that you describe is a known phenomenon, especially in Python.  More information about this is in the link below.

https://github.com/IntelRealSense/librealsense/issues/9134#issuecomment-852000394

Yes, for the plane angle issue an image would be helpful for comparison to similar past cases.  Thanks!

meiy-px, 2021-08-17:
We will test more cameras and then update this issue.

MartyG-RealSense, 2021-08-17:
Thanks very much @meiy-px - I look forward to your further test results.  Good luck!

MartyG-RealSense, 2021-08-29:
Hi @meiy-px  Do you have test results about this case that you can provide, please?  Thanks!

closed 2021-09-09 by MartyG-RealSense