realsenseai/librealsense #9474: D455 no frame received on SDK v2.48.0

YuxuanSnow, 2021-07-23:
Dear Realsense Developers,

i met now an issue with my Realsense d455 camera. When i use the v2.48.0, there is no frame received for RGB and Depth stream in Realsense-viewer. Besides, i launched the code example "hello-realsense" and the program always exits with code 1, because the program waits for frames and apparently no frames received from the camera!

I think the problem is not with privacy setting, because now i can detect the camera in device manager under "camera" category, and i can also open the "camera" application on Windows and it can actually work!

Also something weird is, i can sometimes see both RGB frames and Depth frames in Realsense-Viewer in v2.45.0. Unluckily, it no more works while i was posting this issue.

I am planning to use Realsense SDK to write my own program. Please help me fixing this problem! Thanks.

MartyG-RealSense, 2021-07-23:
Hi @YuxuanSnow  May I first ask if you already have the depth stream enabled in the RealSense Viewer when using **rs-hello-realsense**, or if you have an example RealSense program already running when you enable streams in the Viewer.  

You can only have one program using a particular camera stream at a time, as that program 'claims' access to the stream that has been enabled and stops other programs from using that particular stream until the stream is stopped.

YuxuanSnow, 2021-07-23:
Hi, 
no, I didn't use the two programs at the same time. The v2.48.0 realsense viewer never works, and now I use the v2.43.0, I can see images in realsense-viewer. However, I still cannot use the rs-hell0-realsense. The program hangs at "p.wait_for_frames". So the program exit with 1 (Failure).

YuxuanSnow, 2021-07-23:
Update: also tried with d435i, but same issue as d455. The prebuilt binary file of rs-hello-realsense works fine on v2.43.0, and the realsense-viewer works. However if i build it in Visual Studio 2019, the program hangs at `rs2::frameset frames = p.wait_for_frames();`. I suppose that is because no frame is received for the program.

MartyG-RealSense, 2021-07-23:
The SDK example programs are built in a different way to those created by RealSense users.  All of the examples are built together instead of individually.   If one of the examples are modified then the **make** instruction is issued again but this time only the programs that were modified are updated.  

https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237

This process is described for **rs-hello-realsense** in the link below.

https://github.com/IntelRealSense/librealsense/issues/6797

If you are seeking to learn how to create your own RealSense application with Visual Studio using the SDK's provided **.prop** files then the link below is very useful.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

There is also a **Getting Started** tutorial project in the link below, which does not have dependencies and so is a good learning experience before doing projects with dependencies.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

YuxuanSnow, 2021-07-25:
Hi,

appreciate for your answer. I already tried all your recommondation, but none of them solves this issue. I also tried to use cmake to compile the code, and the `rs-hello-realsense` has result like this:
[image]

I think the issue is still there: the program doesn't receive the frames from d455 or d435i. This is very weird because the "camera" app on windows works fine with the d455. And the binary file of `rs-hello-realsense` downloaded from the github repository works fine. However, if i compiled the code and run the program, it doesn't receive frames anymore. Can you tell me if there is any other possible solution?

MartyG-RealSense, 2021-07-25:
Since the build process for the set of SDK-provided examples is different from how RealSense projects that you have created yourself are built in Visual Studio, you may not necessarily learn much that is applicable to building other projects from getting a build of **rs-hello-realsense** working.  So I would recommend not being too concerned about it.

A good strategy may be to run the pre-built executables, and then look at the cpp source code of those examples at the link below to see how the code could be adapted for your own project scripts.

https://github.com/IntelRealSense/librealsense/tree/master/examples

MartyG-RealSense, 2021-08-02:
Hi @YuxuanSnow  Do you require further assistance with this case, please?  Thanks!

YuxuanSnow, 2021-08-02:
No, i think it is more like a OS uncompatible error. Under Linux it works fine. Thanks anyway for your support!

closed 2021-08-02 by YuxuanSnow