realsenseai/librealsense #9475: L515 depth_frame interference problem

dohyeonYoon, 2021-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                           | v2.47.0                           | 
| Language                              | python                          | 
| Segment			        |  others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
While rendering the color map and depth map with 3 L515 cameras, optical interference occurred in the depth_colormap as shown in the following picture.
Is there a way to eliminate optical interference with only code without HW sync?

thank you.

MartyG-RealSense, 2021-07-23:
Hi @dohyeonYoon   In this particular case the poor depth map may not be related to the presence of multiple cameras, but instead because of how the L515 lidar camera model reacts to highly reflective objects (such as the transparent plastic drink bottle in the image).

With the L515, the light may go straight through the reflective transparent object and be reflected off a surface / object behind the transparent object.  This could result in no depth values being registered for the transparent object because it is not reflecting light back into the camera's receiver.

dohyeonYoon, 2021-07-24:
Then how can I adjust the receiver's reception rate with the api?

MartyG-RealSense, 2021-07-24:
L515 receiver gain and laser power can be reduced by setting the **Short Range** camera preset.  This can be done in the options side-panel of the RealSense Viewer or in scripting with the **RS2_L500_VISUAL_PRESET_SHORT_RANGE** instruction.

**Python**
https://github.com/IntelRealSense/librealsense/issues/8161#issuecomment-761908864

**C++**
https://github.com/IntelRealSense/librealsense/issues/9071#issuecomment-878552273

closed 2021-08-05 by MartyG-RealSense