realsenseai/librealsense #9477: D455 Gain adjust issue, RGB image is blue after adjusting RS2_OPTION_GAIN

BigYuanHead, 2021-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |    Linux Ubuntu 18 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.48.0                  | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description
 RGB image is blue after adjusting RS2_OPTION_GAIN.
The code below is how I adjust the gain for the RGB sensor.
```
// [RGB] configuration RGB sensor gain 
if (RGB_sensor.get_option(RS2_OPTION_GAIN))  
    RGB_sensor.set_option(RS2_OPTION_GAIN, 16);  //16-248
// [RGB] Enable AUTO Exposure
if (RGB_sensor.get_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE))
    RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1); 
```

 And this is how that image looks like.
[
[image]
](url)

MartyG-RealSense, 2021-07-23:
Hi @BigYuanHead  If your RGB image on D455 has a blue tint due to reducing the gain, you could try compensating by increasing the **white balance** value to a value of '5000' or more, as demonstrated in the images below from the RealSense Viewer tool.

[image]

[image]

Below is example C++ code for disabling auto white balance and setting a manual white balance value that I have modified to fit your script.

```
RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, false);
RGB_sensor.set_option(RS2_OPTION_WHITE_BALANCE, 5000);
```

BigYuanHead, 2021-07-24:
> Hi @BigYuanHead If your RGB image on D455 has a blue tint due to reducing the gain, you could try compensating by increasing the **white balance** value to a value of '5000' or more, as demonstrated [...]

Hi,
Thank you for your idea, but after I modified the script as you say, the blue tint is still there.
Here I attach the code after I modified.
```
// [RGB] configuration RGB sensor gain
if (RGB_sensor.get_option(RS2_OPTION_GAIN))
    RGB_sensor.set_option(RS2_OPTION_GAIN, rsParams->RGB_GAIN); //16-248
// [RGB] Enable AUTO Exposure and AUTO White balance
if (RGB_sensor.get_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE))
    RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, false);
RGB_sensor.set_option(RS2_OPTION_WHITE_BALANCE, 6000);
if (RGB_sensor.get_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE))
    RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, true); 
```

MartyG-RealSense, 2021-07-24:
Further testing confirmed that the blue tint remained if auto white balance was not disabled.

Does it make a difference if you use 0 instead of false for RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE

**RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, 0);**

Also, for the line **RGB_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, true); ** did you mean to set it to **false** if you are testing whether it is true in the If statement in the line above it?  Again, I would also suggest trying '0' instead of 'false' for that instruction too.

closed 2021-08-05 by MartyG-RealSense