realsenseai/librealsense #9478: rs2_get_frame_vertices in C

MattCubed, 2021-07-23:
Hi! Im trying to use the L515 in my project and I can not get rs2_get_frame_vertices to work in C. Is there example code somewhere I can view?

MartyG-RealSense, 2021-07-23:
Hi @MattCubed   A point cloud script in C language shared by a RealSense user in the link below provides an example use of **rs2_get_frame_vertices** within it.

```
rs2_vertex* vtx = rs2_get_frame_vertices(points, &e);
check_error(e);

```
https://github.com/IntelRealSense/librealsense/issues/4958#issuecomment-537873138

MattCubed, 2021-07-23:
He creates rs2_vertex* vtx but never uses vtx to get the x,y or z of the readings and when using that example,I cant read out the coordinates either. Thank you for the fast response though

MartyG-RealSense, 2021-07-23:
While not exactly what you are looking for, the C++ script **rs-pcl-color** provides an example of retrieving XYZ from points vertices that might provide inspiration for approaching the problem in your own project.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L118

MattCubed, 2021-07-25:
I think I'm closer but I keep getting the error "Object does not support "librealsense::points" interface!" when I try to run the program. I think its because im getting a normal frame and not a point frame but im not sure how to get that. I tried changing the format to RS2_FORMAT_XYZ32F but when I do, it just prints the error message "This device does not support depth streaming"

MartyG-RealSense, 2021-07-26:
The most common stream formats are as follows:

Depth: Z16
Infrared: Y8
RGB: RGB8
IMU (gyro and accel): MOTION_XYZ32F

MattCubed, 2021-07-26:
Alright, I changed back to what I had it originally (RS2_FORMAT_Z16). I'm getting segmentation faults when trying to access the array within the rs2_vertex structure.

(Using original example provided)
rs2_vertex* vtx = rs2_get_frame_vertices(points, &e);
fprintf(stderr,"%f\n",vtx[0].xyz[0]);

MartyG-RealSense, 2021-07-26:
I researched your question carefully but did not find an obvious solution.

Given that **rs2_vertex** has linkages to rs_types.h:

https://intelrealsense.github.io/librealsense/doxygen/structrs2__vertex.html

Does it make a difference if you add the include below to the include list at the top of your script?

**#include <librealsense2/h/rs_types.h>**

closed 2021-08-08 by MartyG-RealSense