realsenseai/librealsense #9479: How to get the size of PointCloud.ply of YOLO predefined region

yiyangd, 2021-07-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 5.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-77-generic | 
| Platform                                 | PC |
| SDK Version                            |2.44.0 | 
| Language                            | open3d/python | 
| Segment			| others | 

### Issue Description

My goal is capturing a Bottle by running Yolov3  and do the registration with the Bottle modelled by using FreeCAD with the exact bottle size (e.g. 16cm height). 

However, I noticed that the bottle_yolo.ply is significantly smaller than the bottle_CAD.ply. 

[image]

Following is I re-scale bottle_yolo by 500 and bottle_CAD by 0.01.

I would like to know is there any api to replace my manual solution and could re-scale the bottle_yolo to the real bottle size (as what I did with FreeCAD)?

Thanks!

MartyG-RealSense, 2021-07-24:
Hi @yiyangd  In general, the 3D software tool that a RealSense ply file is imported into should be set to the same scale that was used in the RealSense SDK during the ply's creation in order to minimize the chances of distortion occurring due to scaling.  The default depth unit scale for 400 Series cameras in the SDK is **0.001**

yiyangd, 2021-07-27:
However, is there any API that I can clearly get and control both size?

> Hi @yiyangd In general, the 3D software tool that a RealSense ply file is imported into should be set to the same scale that was used in the RealSense SDK during the ply's creation in order to [...]

MartyG-RealSense, 2021-07-27:
Could you provide some more information about your project please to aid my understanding of it.  Do you:

1.  Create **bottle-yolo.ply** first with YOLOv3 and then perform registration on it with FreeCAD to create **bottle-cad.ply** ?  Or do you instead

2.  Create **bottle-cad.ply** first with FreeCAD and then perform registration on it with YOLOv3 to create **bottle-yolo.ply**

Your statement that "I noticed that the bottle_yolo.ply is significantly smaller than the bottle_CAD.ply" suggests to me that you created bottle_CAD.ply first and it was the correct scale, but after you performed registration the resulting bottle-yolo.ply has the wrong scale because it is significantly smaller.  So you re-scale both ply files manually with different scale values until you end up with a result that works for you.

I am not aware of the existence of any API solution that could automate this conversion process between FreeCAD and YOLOv3.

It honestly seems to be an over-complicated process, when you could instead just capture a point cloud of the bottle in the 3D mode of the RealSense Viewer tool, export a ply and then import the ply file into FreeCAD or another free 3D modelling tool such as MeshLab or Blender (making sure to set the depth unit scale in that 3D software to **0.001** to match up with RealSense's depth scale when the original ply was created).

An example tutorial of using the Viewer and MeshLab for ply export and import can be found in the link below.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

MartyG-RealSense, 2021-07-28:
As a new case has been posted at https://github.com/IntelRealSense/librealsense/issues/9504 let's close this one and continue it there.  Thanks!

closed 2021-07-28 by MartyG-RealSense