realsenseai/librealsense #9481: Can't stream RSdevice besides 0 index

MatthewLKomar, 2021-07-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 * 2 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.47.0.3309                          | 
| Language                            |  Unity         | 
| Segment			|             C#       | 

### Issue Description
[image]

In RSDevice.cs any stream index > 0 causes cfg to be effectively null.
The culprit is in line 70  - 71. 
```
using (var cfg = DeviceConfiguration.ToPipelineConfig())
            ActiveProfile = m_pipeline.Start(cfg); 
```
The error code is: rs2_pipeline_start_with_config(pipe:000001B735A2C0B0, config:000001B72011BD20)

I've looked at #2203 https://github.com/IntelRealSense/librealsense/issues/2203
I've looked at #2333 https://github.com/IntelRealSense/librealsense/issues/2333
These both offer no help to this problem. 

Any ideas on how I can go about fixing this?

MartyG-RealSense, 2021-07-24:
Hi @MatthewLKomar  The error **Couldn't resolve requests** means that the RealSense SDK could not provide a requested camera configuration, either because the requested configuration is not currently supported on that particular camera model or because there is a typing error in the stream definition instruction that the SDK cannot understand.

For a multiple camera setup in the RealSense Unity wrapper, the link below advises to use multiple device components with their own unique serial number.

https://github.com/IntelRealSense/librealsense/issues/8347#issuecomment-778745413

Expanding upon this advice, a detailed guide shared by a RealSense user for creating a Unity multicam setup is provided here:

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper

You may also find interesting a recently released project for calibrating together multiple RealSense cameras in Unity using **Apriltag** image tags.

https://github.com/ababilinski/realsense-apriltags-calibration-unity

MatthewLKomar, 2021-07-27:
Hi @MartyG-RealSense, thanks for getting back to me. 
That github link was the motherlode thanks a lot. 

But I have a question: what does StreamIndex do? 
Looking at ababilinski's implementation and fixing mine, I noticed that the serial number needs to change, but the stream index stays constant at -1. Why isn't serial number tied to Stream Index?

MartyG-RealSense, 2021-07-27:
In the RealSense Unity wrapper, **Stream Index** can be used to differentiate more than one stream profile of the same type - for example, having two **Infrared** profiles but giving one of them an index of '1' and the other an index of '2' so that they represent the **left and right** infrared streams.  This is described in a recent Unity case in the link below.

https://github.com/IntelRealSense/librealsense/issues/9427#issuecomment-881515532

Therefore, the Stream Index can define how multiple streams of the same format are handled, whilst the Serial Number will determine which camera in a multiple set of cameras the settings will be applied to.

I have added a **Documentation** tag to this case so that the RealSense team can at a future date consider your suggestion to add the link below to the Community Projects section of the SDK examples page.  This case should be kept open whilst that documentation request is active, but you do not need to take any further action about it.

https://github.com/ababilinski/realsense-apriltags-calibration-unity

MartyG-RealSense, 2026-01-30:
Issue closed due to RealSense no longer updating support for Unity.

closed 2026-10-23 by MartyG-RealSense