realsenseai/librealsense #9483: HW sync between two D435 by pyrealsense

zhangy76, 2021-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version | WIN10 | 
| Platform                                 |  PC |
| SDK Version                            |  pyrealsense2         | 
| Language                            |  python    | 
| Segment			|  others         | 

### Issue Description
My program is simple, HW sync two D435. 

I follow the white paper to connect the cables. Setting the master (1) and slave mode (2) for the two cameras as below

```
ctx.query_devices()[0].first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, 'mode number')
pipe = rs.pipeline(ctx)
cfg = rs.config()
cfg.enable_device(serials[0])
cfg.enable_stream(rs.stream.depth, depth_resolution_width, depth_resolution_height, rs.format.z16, depth_frame_rate)
cfg.enable_stream(rs.stream.color, depth_resolution_width, depth_resolution_height, rs.format.bgr8, depth_frame_rate)

pipe.start(cfg)

frames = pipe.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
timestamp = frames.get_timestamp()
```

However, the slave camera does not follow the frame rate of the master camera. I am not sure if the sync cable is connected correctly (even though the same as intruction) or the program is wrote incorrecly.

MartyG-RealSense, 2021-07-25:
Hi  @zhangy76   It is more accurate to state that the slave camera syncs with the **frame timestamp** of the master camera rather than matching its frame rate.

Assuming that your script is able to successfully set the inter_cam_sync mode of one of the cameras to '1' (master) and the other camera to '2' (slave), the link below provides advice about verifying whether the timestamp of the slave camera is following that of the master camera.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894574-How-to-get-the-same-timestamp-data-from-two-cameras

zhangy76, 2021-07-25:
Thank you very much for the reply.

(1-2) What do you mean by sync with the frame timestamp when using HW sync?

(3) For the Master camera, I just set mode number = 1 in "ctx.query_devices()[0].first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, 'mode number')". I test the command by checking the output of ".get_option(rs.option.inter_cam_sync_mode, 'mode number')" is the same as my input.

MartyG-RealSense, 2021-07-25:
Thanks very much for the details about how you set the sync mode value.

When multiple cameras are hardware synced with each other, the slave cameras that are set to '2' follow the timing of a trigger pulse from a master device (either a camera set to '1' or a signal generator device).  This means that the slave cameras align their frame timestamps to the timing of the master signal.  

If the slave cameras have streams enabled but are not hardware synced successfully then they will still capture but each one will operate independently with no interaction with the master signal or with other slaves.

If the slaves are set to '2 then they will listen for a certain time period for a trigger signal on each frame and then stop listening and capture independently if they do not detect the trigger pulse.

If Inter Cam Sync Mode is set to '4' or higher (Genlock mode) then the slaves will wait indefinitely for a trigger signal and not capture independently no matter how much time passes.

**************

If your aim was to instead ensure that both cameras are running at roughly the same FPS, it may be better to **enforce the FPS** instead of using hardware sync, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

zhangy76, 2021-07-26:
Thanks. I found HW sync is really hard. I test the connection using IntelRealsense viewer. But the sync seems not work at all. I will focuses on enforcing the FPS then.

closed 2021-07-26 by zhangy76