realsenseai/librealsense #9492: how to get callback function of depth、color and IMU separately

oym1994, 2021-07-26:
Hi, Dear developer
I want to handle something in the relative callback function. Now I just example in "pose and image":
`std::string serial;
    if (!device_with_streams({ RS2_STREAM_POSE }, serial))
        return EXIT_SUCCESS;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    if (!serial.empty())
        cfg.enable_device(serial);
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    // Enable both image streams
    // Note: It is not currently possible to enable only one
    cfg.enable_stream(RS2_STREAM_FISHEYE, 1, RS2_FORMAT_Y8);
    cfg.enable_stream(RS2_STREAM_FISHEYE, 2, RS2_FORMAT_Y8);

    // Define frame callback

    // The callback is executed on a sensor thread and can be called simultaneously from multiple sensors
    // Therefore any modification to common memory should be done under lock
    std::mutex data_mutex;
    uint64_t pose_counter = 0;
    uint64_t frame_counter = 0;
    bool first_data = true;
    auto last_print = std::chrono::system_clock::now();
    auto callback = [&](const rs2::frame& frame)
    {
        std::lock_guard<std::mutex> lock(data_mutex);
        // Only start measuring time elapsed once we have received the
        // first piece of data
        if (first_data) {
            first_data = false;
            last_print = std::chrono::system_clock::now();
        }

        if (auto fp = frame.as<rs2::pose_frame>()) {
            pose_counter++;
        }
        else if (auto fs = frame.as<rs2::frameset>()) {
            frame_counter++;
        }

        // Print the approximate pose and image rates once per second
        auto now = std::chrono::system_clock::now();
        if (now - last_print >= std::chrono::seconds(1)) {
            std::cout << "\r" << std::setprecision(0) << std::fixed
                      << "Pose rate: "  << pose_counter << " "
                      << "Image rate: " << frame_counter << std::flush;
            pose_counter = 0;
            frame_counter = 0;
            last_print = now;
        }
    };

    // Start streaming through the callback
    rs2::pipeline_profile profiles = pipe.start(cfg, callback);

    // Sleep this thread until we are done
    while(true) {
        std::this_thread::sleep_for(std::chrono::milliseconds(10));
    }

    return EXIT_SUCCESS;`

And my question is: 
Can we register different callback function for different data type?  The callback function must be a static function, while I want to register to a class member function(non- static), how can I achieve that?

Thank you for your attention!

RealSenseSupport, 2021-08-02:
@oym1994 Sorry I'm afraid you have to use a single callback. But you can dispatch frames to different functions.
In the sample, the callback function is a lambda function. You can call your member function in the lambda function.

RealSenseSupport, 2021-09-17:
@oym1994 Any other questions about this? Looking forward to your reply. Thanks!

oym1994, 2021-09-18:
Not anymore! Thanks for your kind response! The only pity is the bad news that Intel announce to dissolve the Realsense department and no new Realsense products could be bought from the market anymore. It would be better if you could provide more details on this decision. Anyway, thank you!
Best, Slamer

RealSenseSupport, 2021-09-23:
@oym1994 Thanks for your update! We'll continue to provide D400 stereo products.  And if no other questions, close this ticket accordingly.

closed 2021-09-23 by RealSenseSupport