realsenseai/librealsense #9494: What is the depthScale setting in the project2dPixelToDepthPixel method?

c1105590204, 2021-07-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Temporarily unavailable Camera Model and Firmware Version.
android 11,sdk is 2.0, language is java
Descroption:
When I looked at the example, I found that the parameter mDepthScale could not find the initial value, and I didn’t find a way to get it elsewhere.
The code in wrappers/android/code_fragments/UtilsFragments.java

`
private int framesCounter = 1;
        Runnable mStreaming = new Runnable() {
@Override
public void run() {
        try {
            try(FrameSet frames = mPipeline.waitForFrames()) {
                try(FrameSet processed = frames.applyFilter(mColorizer)) {
                    mGLSurfaceView.upload(processed);
                    if((++framesCounter % 80) == 0) {
                        // getting depth frame
                        DepthFrame depthFrame = frames.first(StreamType.DEPTH).as(Extension.DEPTH_FRAME);
                        Intrinsic depthFrameIntrinsic = depthFrame.getProfile().getIntrinsic();
                        // getting depth frame field of view
                        float fov[] = Utils.getFov(depthFrameIntrinsic);
                        Log.d(TAG, "fov = (" + fov[0] + ", " + fov[1] + ")");
                        // getting color frame
                        VideoFrame colorFrame = frames.first(StreamType.COLOR).as(Extension.VIDEO_FRAME);
                        Intrinsic colorFrameIntrinsic = colorFrame.getProfile().getIntrinsic();
                        // computing extrinsics between the profiles
                        Extrinsic depthToColorExtrinsic = depthFrame.getProfile().getExtrinsicTo(colorFrame.getProfile());
                        Extrinsic colorToDepthExtrinsic = colorFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
                        // finding depth pixel with distance not zero (means that there is depth data in this pixel)
                        int w = depthFrameIntrinsic.getWidth();
                        int h = depthFrameIntrinsic.getHeight();
                        byte[] depthFrameData = new byte[ w * h ];
                        depthFrame.getData(depthFrameData);
                        float depthDataAtMiddleOfFrame = depthFrameData[w * h / 2];
                        int halfWidth = w / 2;
                        while (depthDataAtMiddleOfFrame < 0.2) {
                            halfWidth += 10;
                            depthDataAtMiddleOfFrame = depthFrameData[halfWidth * h];
                        }
                        // using get distance api to convert from raw distance to meters
                        float depthAtMiddleOfFrame = depthFrame.getDistance(halfWidth, h / 2);
                        Log.d(TAG, "depth = " + depthAtMiddleOfFrame);
                        // deproject, transform, project - in order to convert from depth pixel to color pixel
                        Pixel depth_pixel = new Pixel(halfWidth, h/2);
                        Point_3D depth_point = Utils.deprojectPixelToPoint(depthFrameIntrinsic, depth_pixel, depthAtMiddleOfFrame );
                        Point_3D color_point = Utils.transformPointToPoint(depthToColorExtrinsic, depth_point );
                        Pixel color_pixel = Utils.projectPointToPixel(colorFrameIntrinsic, color_point);
                        Log.d(TAG, "depthPoint = (" + depth_point.mX + ", " + depth_point.mY + ", " + depth_point.mZ + ")");
                        Log.d(TAG, "colorPoint = (" + color_point.mX + ", " + color_point.mY + ", " + color_point.mZ + ")");
                        Log.d(TAG, "pixel = (" + depth_pixel.mX + ", " + depth_pixel.mY + ")");
                        Log.d(TAG, "color pixel = (" + color_pixel.mX + ", " + color_pixel.mY + ")");
                        // convert color pixel to depth pixel
                        float depthMin = 0.1f;
                        float depthMax = 10.f;
                        Pixel toPixel = Utils.project2dPixelToDepthPixel(depthFrame, mDepthScale, depthMin, depthMax,
                        depthFrameIntrinsic, colorFrameIntrinsic, colorToDepthExtrinsic, depthToColorExtrinsic, color_pixel);
                        Log.d(TAG, "to_pixel = (" + toPixel.mX + ", " + toPixel.mY + ")");
                    }
                }
            }
            mHandler.post(mStreaming);
        }
        catch (Exception e) {
            Log.e(TAG, "streaming, error: " + e.getMessage());
        }
    }
};
`
How do i need to pass this parameter? 

thanks!!

MartyG-RealSense, 2021-07-26:
Hi @c1105590204  The script in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/code_fragments is not designed to be compiled and run, but just to act as 'code snippets' that could be used as a reference for how to program certain functions in Java.

It looks as though the **UtilsFragments.java** script is the only reference that uses the **mdepthscale** variable.

If you wanted to retrieve the depth scale for a RealSense camera then you would not necessarily need a programming instruction to do so though, as the value remains fixed unless deliberately changed in the application and you can therefore use that fixed value in calculations involving the depth scale.  

For example, 400 Series cameras use a default depth scale of **0.001** and SR305 / SR305 is **0.000125**.  The depth units of L515, meanwhile, is **0.000250**

c1105590204, 2021-07-27:
> Hi @c1105590204 The script in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/code_fragments is not designed to be compiled and run, but just to act as 'code snippets' that could be used as a reference for how to program certain functions in Java.
> 
> It looks as though the **UtilsFragments.java** script is the only reference that uses the **mdepthscale** variable.
> 
> If you wanted to retrieve the depth scale for a RealSense camera then you would not necessarily need a programming instruction to do so though, as the value remains fixed unless deliberately changed in the application and you can therefore use that fixed value in calculations involving the depth scale.
> 
> For example, 400 Series cameras use a default depth scale of **0.001** and SR305 / SR305 is **0.000125**. The default depth unit value of L515, meanwhile, is **0.000250**

Thank you.

closed 2021-07-27 by c1105590204