realsenseai/librealsense #9495: An evident Timestamp gap between host and  hardware

oym1994, 2021-07-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |   05.12.13.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-77-generic                                         | 
| Platform                                 | PC |
| SDK Version                            | librealsense 2                          | 
| Language                            | C++                         | 
| Segment			|  Robot                          |

`       //Get timestamp in the callback function
        double hostTimeNow = System::Time::now().toSec();
        double hardwareTime = data.get_timestamp() * 0.001;
        mDriverHealth.mHardwareImageFps = 1 / (hardwareTime - mDriverHealth.mHardwareImageTime);
        mDriverHealth.mHardwareImageTime = hardwareTime;
        mDriverHealth.mHostImageFps = 1 / (hostTimeNow - mDriverHealth.mHostImageTime);
        mDriverHealth.mHostImageTime = hostTimeNow;
       
        //Print timestamp later
         std::cout << "HardwareImageFps: " << mDriverHealth.mHardwareImageFps << std::endl;
        std::cout << "HostImageFps: " << mDriverHealth.mHostImageFps << std::endl;
        std::cout << "HostImageTime: " << mDriverHealth.mHostImageTime << " HardwareImageTime: " << mDriverHealth.mHardwareImageTime << "; Time diff " << mDriverHealth.mHostImageTime - mDriverHealth.mHardwareImageTime << std::endl;
`
output
`HardwareImageFps: 30
HostImageFps: 30
HostImageTime: 1627292572.279579162597656 HardwareImageTime: 1627292572.23203182220459; Time diff 0.04754734039306640625
`

Only color image is enabled and set RS2_OPTION_GLOBAL_TIME_ENABLED to be 1.

The problem is that the timestamp diff(0.047s) between the hardware and host is more than the image creation time interval(0.033s). Is it normal? I think the ideal timestamp diff is 0, am I right?

Thank you!

MartyG-RealSense, 2021-07-26:
Hi @oym1994  Global Time is designed to provide synchronization across multiple cameras.  So it may not be necessary to use it in order to sync host time with the hardware timestamp if you are only using a single camera.

https://github.com/IntelRealSense/librealsense/pull/3909

Could you test with Global Time disabled and see what difference it makes to the result of your calculation please?

oym1994, 2021-07-27:
Thanks for ur kind response！For I want to do time synchronization with other device，such as lidar. My solution is to synchronize those device with the PC first！（which means the timestamps of all device are from PC）. BTW，what do u mean when mentioning metadata support？How to verify whether it is supported in my current device and setting？Thanks again for ur kind help！

MartyG-RealSense, 2021-07-27:
Given that you are aiming to sync with non-D435i devices such as lidar, you could perhaps use the **time_of_arrival** timestamp, like a RealSense user in the link below did when synchronizing a RealSense D415 with a non-RealSense **Stereolabs ZED** depth camera.

https://github.com/IntelRealSense/librealsense/issues/2186

In the link below where a RealSense team member describes the different types of timestamp tha are supported by the RealSense SDK, time_of_arrival is defined as "host clock for the time when the frame buffer reaches Librealsense" 

https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803

********************

In regard to metadata support, there are some types of timestamp that need to have an additional layer of support enabled in order to access their values.  In some cases that support may be automatically included when installing the SDK by certain methods, such as in librealsense **Debian packages** or from source code over an internet connection using the **RSUSB backend** installation method.  

When building from source code on Linux without the RSUSB method, the kernel needs to be **patched** to add support for metadata in the SDK.  

On Windows, metadata support can be added by editing the Windows registry or by the much easier method of enabling it in the RealSense Viewer program.

The SDK documentation link below states that when retrieving metadata with the **get_timestamp()** instruction, the timestamp types **sensor_timestamp** and **frame_timestamp** require metadata support to be enabled.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

You can read more documentation about SDK metadata here:

https://dev.intelrealsense.com/docs/frame-metadata

***********

You can check whether metadata support is present with the SDK instruction ** rs2_supports_frame_metadata** - there is a C++ code example for this in the metadata documentation:

https://dev.intelrealsense.com/docs/frame-metadata#section-metadata-query-api

closed 2021-08-08 by MartyG-RealSense