realsenseai/librealsense #9500: Using DWS Api from CSharp

marcoparenzan, 2021-07-27:
Hello everyone.
As I'm a C# developer, I'm trying using DWS api with Realses 515 from C#.
The solution is here is here: https://github.com/marcoparenzan/IntelRealSense, project TestDWS515. It is a .NET FW 4.7.2 project
My steps:
- converted code from example in chapter 8 from pdf
- created the managed API to invoke DWSSDK.dll- copied in bin directory the content of Intel_RealSense_DWS\RealSenseDWS_SDK content
- copied the licence generated with Intel_RealSense_DWS\RealSenseActivationTool\RealSenseActivationTool.exe tool

Probably there are mistakes, as first call, with dws_create, fails.
Function dws_create is mapped as:
[DllImport("dwssdk", CallingConvention = CallingConvention.Cdecl)]
internal static extern DWS_Status dws_create(DWS_Mode mode, bool multiple_boxes, dws_frameCallback usercallback, dws_statusCallback errorcallback, DWS_Box box_buffer, out int handler); // int& outHandler

The issue is that an exception raises like the dwssdk module was not found.
Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'dwssdk': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
 
I have seen different posts with this message. But again, I have copied the RealSenseDWS_SDK content in the bin\debug folder.

I have used dependency walker on DWSSDK.dll file and I see that the app goes into a loop and when it finishes (after long time) it find the "not found module" error and circular references. So I think that the exception is not about the C# source, but internallin into DWSSDK.dll.

Does someone have any idea?

Thanks in advance.

Marco Parenzan

wluciano, 2021-09-08:
Hi Marco,

I saw your code, and I think you have to specify the full name "DWSSDK.dll" in the DllImport.

[DllImport("DWSSDK.dll", EntryPoint = ...)

In my case I ended up creating my own wrapper in c ++.

My problem was when invoking the dws_measureCallback callback, it always throws me a memory access error when it try to report the detection of a box.

RealSenseSupport, 2021-09-23:
@marcoparenzan Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2021-10-06 by RealSenseCustomerSupport