realsenseai/librealsense #9504: Rescale for Registration CAD model with Yolo Detect Objects

yiyangd, 2021-07-27:
Could you provide some more information about your project please to aid my understanding of it.  Do you:

1.  Create **bottle-yolo.ply** first with YOLOv3 and then perform registration on it with FreeCAD to create **bottle-cad.ply** ?  Or do you instead

2.  Create **bottle-cad.ply** first with FreeCAD and then perform registration on it with YOLOv3 to create **bottle-yolo.ply**

Your statement that "I noticed that the bottle_yolo.ply is significantly smaller than the bottle_CAD.ply" suggests to me that you created bottle_CAD.ply first and it was the correct scale, but after you performed registration the resulting bottle-yolo.ply has the wrong scale because it is significantly smaller.  So you re-scale both ply files manually with different scale values until you end up with a result that works for you.

I am not aware of the existence of any API solution that could automate this conversion process between FreeCAD and YOLOv3.

It honestly seems to be an over-complicated process, when you could instead just capture a point cloud of the bottle in the 3D mode of the RealSense Viewer tool, export a ply from the Viewer and then import the ply file into FreeCAD or another free 3D modelling tool such as MeshLab or Blender (making sure to set the depth unit scale in that 3D software to **0.001** to match up with RealSense's depth scale when the original ply was created).

An example tutorial of using the Viewer and MeshLab for ply export and import can be found in the link below.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

_Originally posted by @MartyG-RealSense in https://github.com/IntelRealSense/librealsense/issues/9479#issuecomment-887711336_

yiyangd, 2021-07-27:
Hi MartyG,

Here is my response with photos

https://docs.google.com/document/d/1sJXL5i-qql0mhJOSjcHi-_ifuk8UNEpbJlcqGD4GQ-s/edit?usp=sharing

yiyangd, 2021-07-27:
Before the scaling, after manual registration, the yolo_bottle is ONLY a blue point inside CAD_bottle.ply

[image]

After yolo_bottle is scaled UP BY 500, CAD_bottle is scaled DOWN BY 1000, they could be at the same size.

[image]

MartyG-RealSense, 2021-07-28:
So you are stitching the two point clouds together in Open3D using ICP point cloud registration **but** the scaling of the two point clouds is wrong, so that the point cloud of the bottle that is produced by YOLO (yolo_bottle) is scaled tiny enough to appear as a blue dot inside the CAD_bottle point cloud.

 In https://github.com/IntelRealSense/librealsense/issues/9479 you state that CAD_bottle is the correctly scaled object (16 cm height).  If CAD_bottle is the correct size, then this would suggest that the scaling of CAD_bottle should be left alone and only yolo_bottle should be re-scaled, so that it is the correct size and has a shape that overlays the shape of CAD_bottle closely.

I hope that the above description is accurate.   If it is not, please do correct me on any errors.  Thanks for your patience!

closed 2021-08-09 by MartyG-RealSense