realsenseai/librealsense #9505: L515 Depth Sensor Flickering after upgrading from 1.4.1.0 to 1.5.8.1

olivetom, 2021-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Ubuntu20  | 
| Kernel Version (Linux Only)    |  5.8.0-59-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.48 }                          | 
| Language                            |  {REalsense-viewer}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Depth Sensor Flickering as in this video: https://drive.google.com/file/d/1XHQWwsWxtELUO2yREpqfNBMou3DoEvhK

MartyG-RealSense, 2021-07-28:
Hi @olivetom  I was not able to recreate the flickering with your L515.  The frame numbers seem to be incrementing normally without pause during streaming and do not reset.  The visual effect resembles the rendering of the graphics being frequently activated and deactivated, as demonstrated by the disappearing and reappearing of the depth scale at the edge of the depth image.

Given that the problem did not previously occur in firmware version 1.4.1.0, would it be possible please to downgrade your firmware to one version older - **1.5.5.0** - to see whether the graphical disruption still occurs?

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

olivetom, 2021-07-28:
@MartyG-RealSense thank you for the advice.
Downgrading to 1.4.1.0 wasn't possible, but downgrading to 1.5.5.0 was successful. But it didn't solved the disruption.
I retried downgrading to 1.4.1.0 (which is the fw I have in my other three L515) but the following error appears. 
[image]
So I'd downgraded to 01.05.04.01 successfully and disruption still occurs :-/ :facepalm: 
I'll report my finding trying to downgrade until disruption disappears.
Regards,

MartyG-RealSense, 2021-07-28:
I would recommend not performing repeated downgrades, as there is a limit of 20 downgrade attempts before you can no longer downgrade and only upgrade to a newer version from that point onwards.

[image]

MartyG-RealSense, 2021-07-28:
My research located a previous case of flickering on the L515, though the problem was only reported for the depth stream in that particular case.  The RealSense user found that the problem did not occur if they used the camera with their own private computer.

https://github.com/IntelRealSense/librealsense/issues/7736

Could you open the debug log in the RealSense Viewer please by left-clicking on the upward triangle icon at the bottom corner of the Viewer and post an image of the messages generated in the log when the flickering is occurring.

[image]

olivetom, 2021-07-28:
> I would recommend not performing repeated downgrades, as there is a limit of 20 downgrade attempts before you can no longer downgrade and only upgrade to a newer version from that point onwards.
> 
> [image]

I know about the limit but I couldn't downgrade directly to 1.4.1.0. Would you @MartyG-RealSense tell me how to do it please?

olivetom, 2021-07-28:
> My research located a previous case of flickering on the L515, though the problem was only reported for the depth stream in that particular case. The RealSense user found that the problem did not [...]

I have only problems with depth sensor. I've attached a capture of the realsense-viewer log for further reference.

[image]

MartyG-RealSense, 2021-07-28:
I do not know of a reason why the original firmware **1.4.1.0** would have a problem with being installed.  If you obtained that firmware from this link:

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

Then you could try downloading the file from this location instead:

https://downloadcenter.intel.com/download/29635/Firmware-for-Intel-RealSense-L500-Product-Family

***************

The log entry about the warning **receiver light saturation, stream stopped for 1 sec** was very helpful, as it enabled me to find a couple of other occurrences of this error, including the one in the link below where the effect is described as 'pulsing' like in your video.  

https://github.com/IntelRealSense/realsense-ros/issues/1750

olivetom, 2021-07-28:
Short range also throws the same flickering.
[image]

This is the error message realsense-viewer throws when trying to downgrade to 1.4.1.0
[image]

Perhaps the most effective way of closing this issue is to give me instructions on how to downgrade successfully to 1.4.1.0.

MartyG-RealSense, 2021-07-28:
I will seek advice from Intel about your case.  Thanks very much for your patience!

olivetom, 2021-07-31:
@MartyG-RealSense any news about this case? Thank you.

MartyG-RealSense, 2021-07-31:
I do not have advice about this case yet.  I will follow up with Intel about it in this coming week.  Thanks very much for your patience.

MartyG-RealSense, 2021-08-01:
Hi @olivetom  Intel have requested that you return your L515 camera to the place that you purchased it from, using the **RMA** (return merchandise authorization) returns process of that supplier.

Could you also provide the serial number of your camera to me, please?  This information can be found by using the **Info** option at the top of the RealSense Viewer's options side-panel to display the camera's information.

olivetom, 2021-08-04:
Hi @MartyG-RealSense returning the camera is not possible because I need it for my daily work. Can I reset to factory somehow?
The serial number of the camera is f0350553.

MartyG-RealSense, 2021-08-05:
A recommendation to RMA return usually means that it is believed that the problem that has been reported cannot be corrected by the user, such as a malfunction with that camera unit's internal hardware.

A simple test that you can perform with L515 is to unplug the USB cable from the camera, turn it around the opposite direction and re-insert it (USB-C cables are two-way insertion on the micro-sized end).  If that test, or trying the camera on a different computer if you have access to one, does not work then returning the camera to the place that it was purchased from at the first opportunity to do so if it is still possible to use the camera for your work until then would be the recommendation.

MartyG-RealSense, 2021-08-16:
Case closed due to no further comments received.

closed 2021-08-16 by MartyG-RealSense

Giggus, 2022-01-12:
Hi, 

I have exactly the same problem. Did someone found a solution?

olivetom, 2022-01-12:
Hi, the solution was to return the camera using the RMA.
I still trying to do the “golden reset” as a last resort solution before throwing it to garbage.

Mauricio

olivetom, 2022-01-12:
> Hi @olivetom As far as I am aware, L515 cameras do not have the feature of resetting to factory defaults with a 'gold reset' like the 400 Series models can.

Thanks for the comment Marty
 It was my last option as I can’t detect the camera neither on windows nor linux.
So now I’m pretty sure that I should dispose the camera. 😔

Mauricio

MartyG-RealSense, 2022-01-12:
@Giggus  Did your problem begin after upgrading the firmware like happened with @olivetom ?

@olivetom   Did the camera become undetectable after putting it in a cool dry place such as a cupboard or drawer for a period of weeks / months and it no longer was detectable after the first time that it was taken out of storage and used again?

MartyG-RealSense, 2022-01-12:
@Giggus  The firmware would not be updated on its own.  It is possible for an update to be casually performed without really thinking about it if an update prompt appears when the Viewer is launched.

Have you tried downgrading the firmware to an earlier version yet?

Giggus, 2022-01-12:
I didn't update it for sure then. 
I didn't try any firmware downgrade since @olivetom didn't have any success in that

MartyG-RealSense, 2022-01-12:
It is worth trying a downgrade anyway just to confirm whether or not it makes a difference, as what doesn't work for one person might work for another.

You can downgrade using the steps below.

1.  Visit the L515 firmware page at the link below and download firmware file **1.5.5.0** (the version that came before 1.5.8.1).

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

2.  Launch the RealSense Viewer.  If your L515 is detected and you can access the **More** option near the top of the side-panel, select the **Update Firmware** option from the drop-down menu.

[image]

3.  A file selection interface will pop up.  Navigate to the **Signed_Image_UVC_1_5_5_0.bin** file at the location on your computer that you downloaded it to and select it.  The firmware update process will then automatically begin to downgrade the camera firmware to 1.5.5.0.

Giggus, 2022-01-13:
@MartyG-RealSense This morning I tried the camera (nothing changed) and the "receiver light saturation, stream stopped for 1 sec" error didn't appear (yet). 
Could it really be an hardware problem?

MartyG-RealSense, 2022-01-13:
@Giggus  If a firmware downgrade did not correct the problem and reversing the orientation of the micro-sized end of the USB cable did not correct it either then you can try disabling the **GLSL** options in the RealSense Viewer settings interface to see whether that makes a positive difference.  https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 has instructions for doing this.

Giggus, 2022-01-13:
I spoke too early and quickly... Got the problem still.. Please look at the video:

https://user-images.githubusercontent.com/4254231/149314746-81b57470-025c-42c6-b067-dcfc19afd91b.mp4

MartyG-RealSense, 2022-01-13:
@Giggus  This is the same problem that @olivetom had and also another RealSense user in https://github.com/IntelRealSense/realsense-ros/issues/1750

In both cases the recommendation was to return the camera to the retailer that it was purchased from.

Giggus, 2022-01-13:
@MartyG-RealSense I tried the firmware downgrade, the cable switch and orientation and the GLSL, Nothing. I'll return the camera... Thanks!

MartyG-RealSense, 2022-01-13:
Okay, thanks very much @Giggus for the update.  Good luck!