realsenseai/librealsense #9506: gpg: keyserver receive failed: No name

konnoeric, 2021-07-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

As I execute 

`sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
`

, the error code below was shown. How can I fix this problem?
Is this because I am using Ubuntu 18.04? According to the table above, this is available only for Linux (Ubuntu 14/16/17). Is there any supported version for Ubuntu 18.04?

I want to use Realsense depth camera D415 on ROS melodic.

error message : 
> Executing: /tmp/apt-key-gpghome.XH2FQUfw6o/gpg.1.sh --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: No name
Executing: /tmp/apt-key-gpghome.cUVDmMoO2U/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

MartyG-RealSense, 2021-07-28:
Hi @konnoeric  This issue is related to a change in the keyserver from keys.gnupg.net to keyserver.ubuntu.com:80.  This transition was taken account of in advance by providing a keyserver command in the SDK installation instructions with two addresses separated by a '| |' symbol.

-keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE **|| sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE**

A small number of RealSense users have found though that this command has not worked for them.  One user's successful solution when encountering the **No name** error was to use the keyserver.ubuntu.com:80 address in both the first and second halves of the command.

https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-873457374

konnoeric, 2021-07-28:
Thank you very much for your kind reply. So in my case, what I should do is just execute

 `sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ` 

isn't it?
The result is shown below. 

> Executing: /tmp/apt-key-gpghome.suI0hjINj5/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

konnoeric, 2021-07-28:
Sorry for further question. I moved on to next steps and installed SDK2.0 but it seems that it's not successfully installed. How can I check that my installation is succeeded?

I followed [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) installed but the ROS driver which requires SDK 2.0  says that 

> CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:44 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.
 Could not find a package configuration file provided by "realsense2"
  (requested version 2.48.0) with any of the following names:
realsense2Config.cmake
realsense2-config.cmake
 Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
 "realsense2_DIR" to a directory containing one of the above files.  If
 "realsense2" provides a separate development package or SDK, be sure it has been installed.
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:48 (message):
 Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

Do you know why is this happening even though I just installed SDK2.0 right now?

konnoeric, 2021-07-28:
Actually the `realsense-viewer` is working according to the screenshot below. So I think the installation is alright.

[image]

MartyG-RealSense, 2021-07-28:
Yes, that looks okay.  Do please let us know if you do encounter any problems.  Good luck!

gorgeousdays, 2021-07-28:
I also meet this problem afternoon,and i solve it just now.
In my opinion,this problem means we can't add the key successfuly.It seems that we can't download the key form the website.So,I created the key in loacl and add it by using `sudo apt-key add spotify.pgp`.This [link](https://unix.stackexchange.com/questions/361213/unable-to-add-gpg-key-with-apt-key-behind-a-proxy) helps me a lot

gorgeousdays, 2021-07-28:
Hello，I also meet this problem afternoon,and I solve it just now.
In my opinion,this problem means we can't add the key successfuly.It seems that we can't download the key form the website.So,I created the key in loacl and add it by using `sudo apt-key add spotify.pgp`.This [link](https://unix.stackexchange.com/questions/361213/unable-to-add-gpg-key-with-apt-key-behind-a-proxy) helps me a lot.

MartyG-RealSense, 2021-07-28:
Thanks very much @gorgeousdays for sharing your method with the RealSense community!

GKCI, 2021-07-30:
I am having the No Name error also.  I used the solution "keyserver.ubuntu.com:80 address in both the first and second halves of the command" which seemed to run okay.  But when the next command in the script executed, there was an error - 
Err:13 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease  403 Forbidden [IP: 52.218.93.114 80].  Looks like this causes the failure to fetch the line in error and says that the repository is not signed.  Can you help me with this error please

gorgeousdays, 2021-07-30:
> I am having the No Name error also. I used the solution "keyserver.ubuntu.com:80 address in both the first and second halves of the command" which seemed to run okay. But when the next command in [...]

I had the same problem and I tried `sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u`,it's useful for me.Hoping it can also solve your problem.

GKCI, 2021-08-03:
Thanks for the help.  I will try this.

magicmomo123, 2021-08-03:
Is this installation method supported for Ubuntu 20.04? If it is not, that might be my problem.

MartyG-RealSense, 2021-08-03:
Hi @magicmomo123  The Debian packages are supported for Ubuntu 20, as described in the link below.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages

They are designed for use with x86 / x64 processor architectures such as desktop and laptop PCs though, not for Arm devices such as Raspberry Pi and Nvidia Jetson.

Jetson has its own package installation process, described in section **5. Register the server's public key** of the Jetson installation guide:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

MartyG-RealSense, 2021-08-09:
Does anyone on this discussion require further assistance, please?  Thanks!

magicmomo123, 2021-08-11:
Hi, yes. I had a brief hiatus getting this to work. How can I install this on a Raspberry Pi 4 then?

MartyG-RealSense, 2021-08-12:
Hi @magicmomo123  If you were installing on Raspberry Pi 4 and using Raspbian OS then a popular installation guide is the **Acrobotic** one in the link below.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

Some Pi users have recommended using a more up to date version of **protobuf** than the one suggested in the guide, such as protobuf 3.14.

If you are using Ubuntu on the Pi 4 with librealsense then it is recommendable to use **Ubuntu Server 20.04**.  

If you need to install ROS on the Pi 4 then the link below provides useful resources for doing so wit Ubuntu Server 20.04.

https://github.com/IntelRealSense/realsense-ros/issues/1724#issuecomment-786508280

MartyG-RealSense, 2021-08-24:
Case closed due to no further comments received.

closed 2021-08-24 by MartyG-RealSense

AliceJames-1, 2022-05-30:
Hi,
I am using the jetson nano, I did the first step for the key and got the same msg. I proceeded to the next step and get this error:
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
Is there anyone who is using the jetson nano with Ubuntu 20.04, that is able to run the realsense and realsense ros packages
Any help/guide is appreciated

MartyG-RealSense, 2022-05-30:
Hi @AliceJames-1  Arm64 packages for Ubuntu 20.04 on computing devices with Arm processor architecture (including Nvidia Jetson and Raspberry Pi) are not currently supported.  Intel plan to introduce Arm64 package support as part of the next generation of the RealSense networking tool when it is released in an upcoming RealSense SDK version.

It is also my understanding that the Jetson JetPack did not officially support Ubuntu 20.04 until the recently released JetPack 5, as described in links at https://github.com/IntelRealSense/librealsense/issues/7722#issuecomment-1072533637

In the meantime, I would recommend installing both the librealsense SDK and the RealSense ROS wrapper from source code instead of packages on Jetson.  Methods for doing this for the SDK are suggested at https://github.com/IntelRealSense/librealsense/issues/10154#issuecomment-1012115083 

Whilst for the ROS wrapper, you should be able to use the **Method 2** instructions.

**ROS1**
https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources
**ROS2**
https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#option-2-build-from-source

iamstarlee, 2024-08-06:
> Hi @konnoeric This issue is related to a change in the keyserver that phased out keys.gnupg.net. This transition was taken account of in advance by providing a keyserver command in the SDK installation instructions with two addresses separated by a '| |' symbol.
> 
> -keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE **|| sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE**
> 
> A small number of RealSense users have found though that this command has not worked for them. One user's successful solution when encountering the **No name** error was to use the keyserver.ubuntu.com:80 address in both the first and second halves of the command.
> 
> **sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE**
> 
> [#9289 (comment)](https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-873457374)

Thanks! It works for me.