realsenseai/librealsense #9507: Can Dot3D Trial export RGBD image

novasky0709, 2021-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.14.50| 
| Operating System & Version |   Win10 | 
| Platform                                 | PC. |

### Issue Description
Hi team, I am using Dot3D Trial and I found that I can export sensor RGB Image and Pose after a scan. I wonder if deep Image can  be exported too ? and how to do that?
Thanks.

MartyG-RealSense, 2021-07-28:
Hi @novasky0709  If you are asking about exporting a depth image from Dot3D, the bottom of the feature comparison table of Dot3D lists that certain versions of it can export color point clouds in ply and pts formats.

https://www.dotproduct3d.com/featuretable.html

For more details, please contact DotProduct, the publisher of the Dot3D software.

novasky0709, 2021-07-29:
Hi @MartyG-RealSense .Thanks for your reply.

MartyG-RealSense, 2021-08-04:
Case closed due to DotProduct being the technical support contact for Dot3D

closed 2021-08-04 by MartyG-RealSense