realsenseai/librealsense #9508: Get distance information per pixel from depth_colormap

dohyeonYoon, 2021-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   {Window 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |            v2.47.0             | 
| Language                            |  python                       | 
| Segment			|  others                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am trying to write a python code that outputs distance information when inputting a pixel position. 

I want to get distance information per pixel in depth colormap like the following picture

[image]

Do you have any similar projects or resources you can recommend?

thank you

MartyG-RealSense, 2021-07-28:
Hi @dohyeonYoon  The Python scripting in the case linked to below may meet your requirements.

https://github.com/IntelRealSense/librealsense/issues/6749

dohyeonYoon, 2021-07-28:
before get the distance information 

RGB photos are 1280x720 and depth photos are 1024x768

the same two pictures
Due to the height difference between the RGB lens and the Depth lens, there is a difference as shown in the following photo.

 Is there a way to match the depth photo with the RGB photo?
[image]
[image]

Also, if the resolution of the RGB photo and the resolution of the depth photo are different,
Are there any problems with matching?

MartyG-RealSense, 2021-07-28:
If you perform a **depth to color alignment** operation in the RealSense SDK then it will automatically adjust for the resolution differences between depth and RGB.

The SDK has an example Python program called **align_depth2color.py** that demonstrates this principle.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

For the L515 camera model, you will need to adjust the depth and color resolutions on the following lines though to match the depth and color resolutions that you want to use (1024x768 depth and 1280x720 RGB).

**Depth**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L37

**RGB**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L40

dohyeonYoon, 2021-07-29:
thanks for kind reply !
actually I mean, 
I want to match a 1280x720 RGB photo with a 1024x768 depth photo that already saved in my computer 
is it possible ?

MartyG-RealSense, 2021-07-29:
If you are able to create new files then you could use the Python script in the link below to save color to your computer as PNG and depth as an array of scaled matrices saved as an **.npy** file and then align them.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

 If you need to use color and depth PNG images then you may be able to perform the alignment in Python with the SDK's **software_device** function.

https://github.com/IntelRealSense/librealsense/issues/1274

Using software_device with Python can be problematic as it was originally designed for C++, though a ReaslSense user created a 'patch' for Python in the link below to fix issues.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000934242/comments/4403012747923

closed 2021-08-10 by MartyG-RealSense