realsenseai/librealsense #9511: L515 is not recognized anymore after firmware update

TommasoBendinelli, 2021-07-28:
Hello,

After I have updated the camera firmware, it has stopped working. 
If I do rs-enumerate-devices, I get "No device detected. It is plugged in?". 

The device is seen by the system as a USB 2. This is the output if I run "lsusb -D /dev/bus/usb/001/005"
Device: ID 8086:0b64 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64

I am on:
Linux l7142 5.4.0-80-generic #90~18.04.1-Ubuntu SMP Tue Jul 13 19:40:02 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

MartyG-RealSense, 2021-07-28:
Hi @TommasoBendinelli  Which firmware version did you update your L515 camera to, please?

TommasoBendinelli, 2021-07-28:
The latest, I do not know precisely because I cannot use the camera anymore

MartyG-RealSense, 2021-07-28:
The latest L515 firmware at the time of writing this would be **1.5.8.1**. obtained from the page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

Did you perform the firmware update using the **RealSense Viewer** program please?

TommasoBendinelli, 2021-07-28:
Yes

MartyG-RealSense, 2021-07-28:
The camera is being detected by lsusb as an L515.  The '0b64' reference in the first line of your log - **Device: ID 8086:0b64 Intel Corp** -  is the ID number of the L515 camera model, as shown in the list of model ID numbers in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/src/l500/l500-private.h#L22

Was the firmware update performed on a USB 3 connection?  Although L515 now supports USB 2 after an update in one of its firmware versions (it was originally a USB 3 only camera), the firmware update that adds that capability needs to be performed on USB 3 first.

[image]

TommasoBendinelli, 2021-07-28:
was using the camera via USB-C port when I clicked on "update firmware". The camera with this port was working good.

MartyG-RealSense, 2021-07-28:
I would recommend performing firmware updates with the USB-C cable that was supplied with the camera if possible (the 1 meter cable with a full size USB-A connector on the computer end of it).

TommasoBendinelli, 2021-07-28:
Would be nice if there was a warning about in before the update. What can I do now to fix it?

MartyG-RealSense, 2021-07-28:
There was a past case in the link below where an L515 was not detected with rs-enumerate-devices after updating the firmware in the Viewer.  A RealSense support team member at Intel Support recommend using the recovery mode of the **rs-fw-update** firmware update tool, and that resolved the problem for that particular RealSense user.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000370121/comments/1500000332922

rafaelspring, 2021-07-29:
Not sure if this is helpful, but we have seen this with some of our customers as well. In our case on Android ( Samsung Galaxy S20 FE).
L515s that worked perfectly fine in USB 3.2 mode now suddenly would only connect as USB 2.1 devices after a while (customers may or may not have updated FW, but I'd say it's likely).

MartyG-RealSense, 2021-07-29:
Thanks very much @rafaelspring for sharing your experience!

closed 2021-08-10 by MartyG-RealSense