realsenseai/librealsense #9515: Cannot change startup resolution through launch file or nodelet.launch.xml

frankSARU, 2021-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |               Ubuntu 18.04                           | 
| Platform                                 | PC |
| SDK Version                            |  legacy    | 

### Issue Description
I was using my D435i in a review project and it can run properly few months ago.
At that time, when I start my D435i it shows a right resolution which is 640*480.
These days when I want to review this project, my console tell me the input data is not right so I check the problem and found that is caused by resolution of camera. I see that my camera is running on default resolution which is 1280*720 and depth resolution 848*480. I have tried to modify rs_camera.launch file and nodelet.launch.xml like this but no use.
[image]
[image]
[image]

This problem also arises in realsense_viewer, every time I change the resolution of my camera, and next time when I start it , it will change back to default resolution.
I'm so confused now.

MartyG-RealSense, 2021-07-29:
Hi @frankSARU  In the RealSense Viewer it is normal for resolution settings to return to the default settings for that camera model when the Viewer is closed and then re-launched.

In the RealSense ROS wrapper, the system for setting a custom camera configuration during roslaunch changed from wrapper version 2.2.22 onwards.  You must now specify three factors in a custom configuration - stream width, stream length and stream FPS.  If you do not do so then the custom configuration is decided by the wrapper to be invalid and it instead automatically applies the default configuration of the camera model that is being used.  For the D435i, that default configuration is 848x480 depth and 1280x720 color.

Below is an example of defining a custom configuration for depth and color in a roslaunch under the new system.  You need to provide width, length and FPS for each stream type that you are using.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

Under the new post-2.2.22 system, numerous settings including FPS are defined in the rs_camera.launch file as '-1', meaning that unless you provide FPS in the custom configuration then it will consider your custom configuration invalid and instead use the camera's defaults.

[image]

frankSARU, 2021-07-29:
Thanks a lot. I just knew this new feature. That works.

closed 2021-07-29 by frankSARU