realsenseai/librealsense #9516: Color and depth image alignment in VB.NET

redyoung75, 2021-07-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 2.45 ? | 
| Operating System & Version |   {Win (10)| 
|                                    | 
| Platform                                 | PC
| SDK Version                            |  { legacy / 2.4.7 }                          | 
| Language                            |  VB.NET                     | 
| Segment			|  others                          | 

### Issue Description: question 

I am new github so I am not sure I am doing the right thing..
Anyway, I am using very unusual VB.NET (because I am used to this) to acquire color and depth images from D435i and D455. I can acquire images but they are not aligned. But I noticed there is a built-in function in SDK for image alignment and I saw many image alignment examples for python and c but none for VB.NET. Is there VB.NET examples for image alignment that I can learn from it?
Thank you for your help!

MartyG-RealSense, 2021-07-29:
Hi @redyoung75  Of the VB.NET programming references that I am aware of, the best one that I know of for depth to color alignment is the link below.

https://github.com/IntelRealSense/librealsense/issues/1983#issuecomment-406866163

[image]

Other VB.Net users reading this case in future may also be interested in the list of RealSense VB.NET resources that I posted on your other question at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360053003673/comments/4404430783635

redyoung75, 2021-07-29:
Thanks MartyG for this.. Maybe I didn't search hard enough... By the way, I tested this by copying original post and ran in VB.Net 2017 and I keep getting error at align.process... The only difference is I am running this in Windows Form App.
Anyway, although aligned and frames are Frames etc, I am getting:
System.invalidcastexception and it says one is Frame and the other is Frames etc which doesn't make sense to me....
Can you provide some guide here?

MartyG-RealSense, 2021-07-30:
I am not familiar enough with C# and VB.Net programming to advise on this question myself, though I did find a C# case where System.invalidcastexception was occurring when using align.

https://github.com/IntelRealSense/librealsense/issues/1013

redyoung75, 2021-07-30:
Thanks MartyG for your help. 
I found out the solution for this... In align.process, there are three seperated functions. Instead of using one that uses a single frameSet as variable, use one with FrameSet and FrameReleaser. The error disappears and alignment function starts working. But you must dispose aligned frame each time otherwise you will eventually run into error.
By the way, will you be able to provide some explanation on what the framereleaser is doing?
Thanks..

MartyG-RealSense, 2021-07-30:
Advice on the subject of frame release in .NET is provided here:

https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923

redyoung75, 2021-08-01:
Thanks MartyG!

closed 2021-08-01 by redyoung75