realsenseai/librealsense #9520: C/C++ transfer ownership of frame data to user managed container

Darustc4, 2021-07-29:
In my experience working with the Realsense SDK in multithreaded applications I have found that the underlying system determining when a frame should be released or not is unreliable at best. When you store frames to be processed in a pipeline, they might get freed and result in a SEGFAULT, even if you call .keep() on them, and this has been an ongoing problem for many, many months.

I wanted to know if it is possible to transfer the data owned by the rs2::frame class to another container I can manage myself. I have had positive results by copying the color and depth data (by doing a memcpy on the data returned by the frame's .get_data() method), but this has an overhead of a few millis of copying so many bytes of data each frame.

Is there any other solution I am missing?

MartyG-RealSense, 2021-07-29:
Hi @Darustc4   Using **software_device** or a **custom processing block** to create custom framesets might meet your needs, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/5847

Darustc4, 2021-07-29:
That is some great guidance, thanks for the support. Closing the issue for now and will reopen if I find problems with it.

closed 2021-07-29 by Darustc4

kayychou, 2023-11-08:
Hi @Darustc4 
Did you manage to transfer ownership of frame data using the suggested methods mentioned above ?
If you have an example that illustrates how you did this, I would be grateful if you could share it with us

I need to take the ownership of the data returned by rs2::frame::get_data() to not be dependant on the rs2::frame lifetime, without making a deep copy.

@MartyG-RealSense have you ever considered adding to the rs2::frame API a method that would transfer the ownership of the data ?
```
std::vector<unsigned char> rs2::frame::transfer_data()
{
    // using a std::move on the std::vector<byte> data owned by the internal frame class in src/frame.h
    return rs2_transfer_data(frame_ref);
}
```

MartyG-RealSense, 2023-11-08:
@kayychou   Custom processing block and software-device remain the two main methods of configuring a custom frameset instead of using the default rs2::frame.  I am not aware of future plans to introduce further custom frameset procedures.

kayychou, 2023-11-08:
Thanks a lot for the support @MartyG-RealSense 

I must be missing some key point but I don't understand how to get ownership of the data acquired by a connected Intel camera using Custom processing block or software-device. Do you have example that illustrates how a custom frameset allows to transfer the ownership of its internal data ?

My goal at the end is to no longer use rs2 types while having the internal frame's `std::vector<byte> data` still alive in memory (avoiding a deep-copy using std::memcpy)
```
// a container used outside of realsense scope, with no more reference to rs2::frameset/rs2::frame
MyContainer data_container;

{
	// Intel D415 connected to USB 3.x port
	rs2::pipeline pipe;
	auto profile = pipe.start();
	rs2::frameset frameset = pipe.wait_for_frames();
	data_container.set_data(...) // taking the ownership of the rs2::frameset internal color data
}

// still able to access the color frame data bytes acquired by the Intel camera (while rs2::frameset has been destroyed)
auto data = data_container.get_data();
```

MartyG-RealSense, 2023-11-08:
There is very limited information available about custom framesets.  https://github.com/IntelRealSense/librealsense/issues/5565#issuecomment-571110857 might be a helpful reference about rs2::frame though, especially in regard to how frames can be stored in memory with the **Keep()** function.

Darustc4, 2023-11-08:
I also did not find any real information on how to use custom frame containers. The keep(] function does not work and you will still get random fatal crashes regardless of how many times you call it per frame.

Your options are either deep copying, figuring out the custom frames through sweat and tears, or abandoning the realsense (as I did).

I know it's a harsh comment, but I seriously can't believe they still have not addressed this in all this time...