realsenseai/librealsense #9523: D455 RGB focal lengths are constantly changing

rafaelspring, 2021-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version |  Win10 | 
| SDK Version                            | 2.47.0 | 

### Issue Description
Our D455 color camera focal lengths are constantly changing. We're using the `video_stream_profile::get_intrinsics()` method to query them.
For a color resolution of 1280x720 we are seeing values in the range of 634.84 (for a cold sensor, just started) to 639.11 (for a sensor that has run for a couple of minutes). Disconnecting the sensor and letting it cool down seems to "reset" the values to their lower end again.

Is this expected behavior? Is this some kind of thermal correction adjusting the focal length values behind the scenes? Is this something we could turn off?

RealSenseSupport, 2021-07-29:
Yes. There is thermal compensation mechanism which adjusting the focal length value. This is to maintain the depth accuracy. We don't want to turn it off.

rafaelspring, 2021-07-29:
@RealSenseSupport Thanks. The changing focal length is in the RGB camera though, not the depth. The depth focal length stays constant.

RealSenseSupport, 2021-08-02:
@rafaelspring Our ENG team confirmed that RGB parameter changing is expected. It is designed for the thermal compensation. Could you please elaborate that why you want this be turned off? We don't have this option now.

rafaelspring, 2021-11-22:
@RealSenseCustomerSupport All good. Feel free to close.

closed 2022-01-05 by RealSenseCustomerSupport