realsenseai/librealsense #9524: metadata not available[ERROR] [1627575249.042885701]: metadata not available

rolandgvc, 2021-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0-77-generic                                         | 
| Platform                                 | Intel NUC |
| SDK Version                            |  2.48                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
After updating the camera to 05.12.14.50, accessing the metadata of a frame through the C++ SDK results in an error. This seems to only happen on my Intel NUC10i7FNK.
```
// config realsense
rs2::pipeline rs_pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 0, RS2_FORMAT_Z16);
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
rs2::pipeline_profile rs_profile = rs_pipe.start(cfg);

rs2::frameset fs = rs_pipe.wait_for_frames();
rs2::depth_frame df = fs.get_depth_frame();
std::cout << df.get_frame_metadata(RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE) << std::endl;

```

MartyG-RealSense, 2021-07-29:
Hi @rolandgvc  The SDK's frame metadata documentation suggests checking that metadata is supported using the **rs2_supports_frame_metadata** instruction in an If condition before retrieving the metadata with **get_frame_metadata** in order to help to avoid generating an exception.

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#librealsense2-api-for-metadata

rolandgvc, 2021-07-29:
Looks like none of the depth frames support the metadata. Thoughts?

MartyG-RealSense, 2021-07-29:
It is possible that metadata support is not enabled on the librealsense SDK build on your NUC.  If you built the SDK with Debian packages in Ubuntu then metadata support is built into the packages.  Metadata support is also included if you build from source code with the **RSUSB backend** method.  If you build the SDK from source code without using RSUSB then metadata support is added by patching the Linux kernel.

The SDK documentation states that **RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE** has been deprecated and replaced by **RS2_FRAME_METADATA_FRAME_EMITTER_MODE**

https://intelrealsense.github.io/librealsense/doxygen/rs__frame_8h.html#a91f19a01f5bf2abadc30959a8d3109c9a79ae181e2bbabecbab365e36575c89bc

rolandgvc, 2021-07-30:
I installed the binaries from apt. Again, everything was working fine before updating the camera software. Would there be a way of downgrading the camera back to its previous version?

MartyG-RealSense, 2021-07-30:
Once you have uninstalled all previous librealsense packages with a purge instruction:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

You can install specific package versions instead of the latest version by providing details of the package name, like in the example guidance in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1925#issuecomment-858716097

rolandgvc, 2021-07-30:
What about camera firmware only?  I think you can do that from realsense-viewer directly, no? Where could I find a list of the versions?

MartyG-RealSense, 2021-07-30:
You can install a specific firmware version by downloading it as a **.bin** file from the list on the firmware releases page.  

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

Then go to the **More** option near the top of the Viewer's options side-panel and select the **Update Firmware** option from its drop-down menu.  This will cause a file selector window to pop up.  Select the firmware .bin folder from the location on your computer that you downloaded it to and then the firmware update will commence.

[image]

rolandgvc, 2021-07-30:
Reverted camera / sdk back to the 2.45 binaries. For the same code and sdk version, the metadata is accessible on my Lenovo T480s and not on NUC. Are there any issues with realsense on NUC specifically that are causing this?

MartyG-RealSense, 2021-07-30:
I note from the specification at the top of this case that you are using Ubuntu 20.04 with kernel 5.4 on your NUC.  My understanding is that among RealSense users, **5.8** is a preferred kernel for 20.04.

MartyG-RealSense, 2021-07-30:
If you left-click on the frame metadata option on a stream panel in the RealSense Viewer on your NUC to overlay the metadata display, does it indicate that per-frame metadata is not supported?

[image]

rolandgvc, 2021-08-01:
> If you left-click on the frame metadata option on a stream panel in the RealSense Viewer on your NUC to overlay the metadata display, does it indicate that per-frame metadata is not supported?
> 
> [image]

It doesn't indicate that, so it should work...

MartyG-RealSense, 2021-08-01:
The next test that you could run with your script is to substitute in **RS2_FRAME_METADATA_TIME_OF_ARRIVAL** in place of RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE, since if TIME_OF_ARRIVAL can be displayed in the Viewer's metadata overlay then you should be able to access it with your script too.

closed 2021-08-13 by MartyG-RealSense