realsenseai/librealsense #9529: D435i使用USB 3.0偶尔被识别成USB type 2.1

meiy-px, 2021-07-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Newest) | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |              C++             | 
| Segment			|  {Robot }                          | 

###
使用D435i的过程中出现两种情况：
1、正常使用时小概率出现USB 3.0被识别成USB type 2.1，导致D435i的分辨率和帧率下降；
2、初次使用时出现USB 3.0被识别成USB type 2.1，需要多次重启后才能正常识别USB 3.0。

导致该问题的原因是什么？
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-07-30:
Translation: D435i using USB 3.0 is occasionally recognized as USB type 2.1

There are two situations in the process of using D435i:

1. There is a small probability that USB 3.0 is recognized as USB type 2.1 during normal use, which causes the resolution and frame rate of D435i to drop;
2. USB 3.0 is recognized as USB type 2.1 when it is used for the first time, and USB 3.0 can be recognized normally after multiple restarts.

What is the cause of this problem?

*************

Hello @meiy-px  May I ask some questions, please.

1.  Are you using the offical USB cable that was supplied with the camera or a cable that you have chosen yourself?

2.  Is the camera plugged directly into the computer or is it attached to a USB hub?

你好@meiy-px 我可以問一些問題嗎？

1. 您使用的是相機隨附的官方 USB 數據線還是您自己選擇的數據線？

2. 相機是直接插入電腦還是連接到USB集線器？

meiy-px, 2021-07-30:
Hello @MartyG-RealSense 

1、Are you using the offical USB cable that was supplied with the camera or a cable that you have chosen yourself?
==》We are using the official USB cable supplied with the camera.

2、Is the camera plugged directly into the computer or is it attached to a USB hub?
==>Both of the two use cases come up against this problem. When the camera is attached to a USB hub, the peoblem happens more frequently.

MartyG-RealSense, 2021-07-30:
Thank you very much for the confirmation that you are using the official USB cable.

Do you have a situation sometimes where the camera starts on USB 3.2 and then changes to USB 2.1 during streaming?

meiy-px, 2021-07-30:
Once it is recognized as USB 3.2, it will not change to USB 2.1. The problem heppens only when camera restarts.

MartyG-RealSense, 2021-07-30:
If you are writing your own application then you could try inserting a **hardware_reset()** instruction into the script to reset the camera when the program is launched.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

meiy-px, 2021-07-31:
What is the main reason of it? Is the connector not tight enough?

MartyG-RealSense, 2021-07-31:
As you mention whether the connector is tight enough, is your camera attached to a moving vehicle or a mobile robot where the USB cable may get moved around or vibrated during motion?

meiy-px, 2021-08-02:
Yes, the camera is attached to a moving robot. But the cable is fixed to the body. And when we start the camera, the robot is motionless.

MartyG-RealSense, 2021-08-02:
Thanks for the confirmation.  I asked about motion because of the possibility that it could cause the USB connector to work its way partially out of the USB port.  This could cause the camera to disconnect and then reconnect in USB 2.1 mode.  Using a screw-lock USB cable and screwing the screw-threads either side of the connector into the pair of small holes either side of the camera's USB port can keep the connector tightly fixed in the port.

https://github.com/IntelRealSense/librealsense/issues/3118#issuecomment-456392939

As mentioned above in https://github.com/IntelRealSense/librealsense/issues/9529#issuecomment-889929224 it would be worth inserting a **hardware_reset()** instruction into the start of your C++ script to reset the camera when the program starts up whilst the robot is not in motion and see if that resolves the problem of the camera connection being mis-detected as USB 2.1.

Grodfer, 2021-08-03:
I think you should change a new type-c usb3.0 than you may solve this problem

meiy-px, 2021-08-03:
> I think you should change a new type-c usb3.0 than you may solve this problem

Are there any requirements of the cable if we don't use the cable supplied with the camera?

MartyG-RealSense, 2021-08-03:
Hi @meiy-px   If you are planning to use a cable length of 2 meters or longer then a high quality cable (also known as 'industrial grade') is important due to the high volume of data that a RealSense camera can transmit through the cable.  

Suppliers of cables that have been demonstrated to work with RealSense 400 Series cameras include **Newnex** and **OCP**.

https://www.newnex.com/realsense-3d-camera-connectivity.php
https://www.ocp.com/product/usb-3-0-3-1-cable-type-c-to-a-gen1-5g-black-intel-d435-realsense-camera-62-00214/

If you require a very long cable length then using cables with **active repeater** components in them to boost the signal is recommendable.  Newnex supplies these cables, and also fiber-optic equipment for extending up to 100 meters.

https://www.youtube.com/watch?v=GLQgR1jT04M

meiy-px, 2021-08-04:
> Hi @meiy-px If you are planning to use a cable length of 2 meters or longer then a high quality cable (also known as 'industrial grade') is important due to the high volume of data that a RealSense [...]

OK， Thx

closed 2021-08-04 by meiy-px

4760lky, 2025-05-21:
My d435i camera seems to be damaged after successfully connecting once. It is connected to the USB 3.0 interface of Lenovo Thinkpad. This situation happened three times occasionally. After the damage, the realsense viewer could not connect to the device. I also changed the computer to a robot industrial computer, and it also broke. Is it because the power supply level of USB 3.0 is unstable?

MartyG-RealSense, 2025-05-21:
Hi @4760lky  A RealSense camera should not be physically damaged by connecting it to a USB port.  If the power being provided to the camera by the port is not sufficient then the camera simply will not connect and be detected.

It is recommended that USB 3.1 is used.  The information below is from the D435i's page in the official RealSense Store.

[image]