realsenseai/librealsense #9530: How to get distance of two points from PointCloud

hn24, 2021-07-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC|
| Language                            |  C#                           | 

### Issue Description
Hi, i'm trying to work with the PointCloud object by this sample code and measure the distance between objects
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support

Any one know how to get the distance between any two points get from PointCloud ?

Thanks

MartyG-RealSense, 2021-07-30:
Hi @hn24  A RealSense user wrote a C# conversion of the SDK's **rs-measure** example program for measuring between two points.

https://github.com/IntelRealSense/librealsense/issues/2809#issuecomment-444502830

The final version of their script can be downloaded as a zip file from the link below.

https://github.com/IntelRealSense/librealsense/files/2666705/RS-DevProjectC.zip

hn24, 2021-07-30:
Thank MartyG,
I will download and play with the sample.

closed 2021-07-30 by hn24