realsenseai/librealsense #9531: Export mesh with color texture using save_to_ply() function

MrTitusZ, 2021-07-30:
-------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415                                                      | 
| Firmware Version                   |  05.12.10.00                                            | 
| Operating System & Version |   Windows 10                                          | 
| Platform                                 |  PC                                                          |
| SDK Version                            |  2.48.0.3381                                            | 
| Language                               |  python                                                   | 
| Segment			         |  others                                                    |

### Issue Description

I was trying to replicate the "export_ply_example.py" from the python examples: https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/export_ply_example.py

My gole is to export a ply file with color texture from one frame.
In the example file it uses rs.colorizer() fuction witch using the depth cam texture.
In line 27 it is saying that "texture can be obtained from color or infrared stream".
I tried to map the color frame to the point cloud, i wrote the documentation and tried to implement the c++ pointcloud example , but nothing works.

How can I get the color texture and how can i assign it to the point clod properly?

MartyG-RealSense, 2021-07-30:
Hi @MrTitusZ  Exporting a ply with color in Python has been shown to be problematic in past cases.  If you are okay with exporting a color ply without normals then the the **export_to_ply** Python script in the link below may suit your needs.

https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

MrTitusZ, 2021-07-30:
Hello @MartyG-RealSense! I am aware of the **export_to_ply** function. My problem with the export_to_ply function is that with it I lose the option to set the encoding type and the normal calculation witch are mandatory for me.

MartyG-RealSense, 2021-07-30:
Any investigation about saving a color ply with Python always ends up coming back to the export_to_ply example linked to above unfortunately, as it is the only confirmed example that works.  This is demonstrated by the save_to_ply example script in the link below, which works but does not export color.

https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152

MartyG-RealSense, 2021-08-06:
Hi @MrTitusZ  Do you require further assistance with this case, please?  Thanks!

MrTitusZ, 2021-08-09:
Hello, if there is no solution to give a texture from a colored frame to the save_to_ply() function, then I do not require further assisstance.
Thank you for your help! :)

MartyG-RealSense, 2021-08-09:
I researched this case again carefully but still could not find any further resources that would enable creating a color ply in Python with save_to_ply instead of export_to_ply.  So I will close the case.  Thanks very much for your understanding!

closed 2021-08-09 by MartyG-RealSense