realsenseai/librealsense #9532: How can I get gyro and accel values from L515 in C#?

asheesh1996, 2021-07-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   {Win 10}                                           | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.47.0 }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot}                          | 

How can I get gyro and accel values from L515 in C#?

MartyG-RealSense, 2021-07-30:
Hi @asheesh1996   The link below contains references for accessing IMU streams in C# using **callbacks** or through the RealSense SDK's **Low Level API**.

https://github.com/IntelRealSense/librealsense/issues/8746#issuecomment-814030029

There is also information in the link below about accessing the IMU through Unity, which is C#-based.

https://github.com/IntelRealSense/librealsense/issues/5993#issuecomment-596358982

MartyG-RealSense, 2021-08-06:
Hi @asheesh1996  Do you require further assistance with this case, please?  Thanks!

asheesh1996, 2021-08-11:
HI @MartyG-RealSense, I found the solution for my issue, after receiving the frames (https://github.com/IntelRealSense/librealsense/issues/8746#issuecomment-814265930) I had to process them to get rotations in Euler then converted it to Quaternions  ([link to example in c++](https://github.com/IntelRealSense/librealsense/tree/master/examples/motion)) 
Thanks for help

closed 2021-08-11 by asheesh1996

MartyG-RealSense, 2021-08-11:
Great to hear that you found a solution, @asheesh1996 - thanks very much for the update and for sharing the details of your solution with the RealSense community  :)