realsenseai/librealsense #9533: How to Detect Frame Drops

tarekmuallim, 2021-07-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |  Ubuntu 18  | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC   |
| SDK Version                            |  2.40.0                      | 
| Language                            | c++                       | 
| Segment			|                         | 

When trying to get a new frame, how to detect if frames drops happen?

Applying something like:  
`rs2::frameset frames =   pipeline.wait_for_frames();`
I need to detect if frames drops happen during this.

MartyG-RealSense, 2021-07-30:
Hi @tarekmuallim  Information on how the RealSense SDK calculates frame drops is provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/7488#issuecomment-704124850

tarekmuallim, 2021-07-30:
@MartyG-RealSense thank you for the information.

That source code help me.

I am wondering, is there any way to get information about if incomplete frame is received?

MartyG-RealSense, 2021-07-30:
There is an SDK instruction called **RS2_NOTIFICATION_CATEGORY_FRAME_CORRUPTED**.  

https://intelrealsense.github.io/librealsense/doxygen/rs__types_8h.html#a5b8efd94f1165be36acea964bf3bdd8dab6d76d2e2b73f5157afbd9a60dcfe1e8

Example scripting for accessing this instruction to detect incomplete frames is suggested here:

https://github.com/IntelRealSense/librealsense/issues/3189#issuecomment-460282633

tarekmuallim, 2021-08-02:
Hi @MartyG-RealSense thanks for the information.
It helped me.

closed 2021-08-02 by tarekmuallim

MartyG-RealSense, 2021-08-02:
I'm pleased that the information was useful to you - thanks for the update!