realsenseai/librealsense #9534: save_to_ply / export_to ply not giving always the same output

pelluchon, 2021-07-30:
Hi, 

I am saving an output of a bag video into ply using the code below. If i am running several times the script, I do not have the same export every time. 

`pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file(input_rs)
profile = pipe.start(cfg)
pc = rs.pointcloud()
colorizer = rs.colorizer()

while True:
    frameset = pipe.wait_for_frames()
    align = rs.align(rs.stream.color)
    frameset = align.process(frameset)
    colorized = colorizer.process(frameset)
    color_frame = frameset.get_color_frame()
    depth_frame = frameset.get_depth_frame()
    pc.map_to(color_frame)
    ply = rs.save_to_ply(inter_output)
    ply.set_option(rs.save_to_ply.option_ply_binary, False)
    ply.set_option(rs.save_to_ply.option_ply_normals, False)
    ply.set_option(rs.save_to_ply.option_ignore_color, False)
    ply.set_option(rs.save_to_ply.option_ply_threshold, True)
    ply.process(frameset)`

I have tried with save_to_ply and export_to ply using the code below for export_to_ply.

`points = pc.calculate(aligned_depth_frame)
    points.export_to_ply(inter_output, color_frame)`

Does anyone had the same issue and if so how to solve it ?

Many thanks

|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 2.41.02657                                              | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                           |
| Language                               |  python                                                   |

MartyG-RealSense, 2021-07-30:
Hi @pelluchon  As a first step, I would recommend adding the term ** repeat_playback=false** to the brackets of the **cfg.enable_device_from_file** instruction.  This will prevent the bag file from looping around to the start when it reaches the final frame during playback.

https://github.com/IntelRealSense/librealsense/issues/7067#issuecomment-671017676

The cfg.enable_device_from_file line of your script would therefore look like this after the addition:

**cfg.enable_device_from_file(input_rs, repeat_playback=false)**

pelluchon, 2021-08-01:
Hi @MartyG-RealSense, Thank you for the quick answer.

I did try and it slighly better. If before, I had one different over 3, now it is more one over 5. That with save_to_ply, with export_to_ply it does not change the issue.

Any other ways ?

Many Thanks

MartyG-RealSense, 2021-08-01:
Is there any further improvement if you add two lines below your pipe start instruction to configure **set_real_time** for playback as false.

```
profile = pipe.start(cfg)
playback = profile.get_device().as_playback()
playback.set_real_time(False)
```

pelluchon, 2021-08-01:
yes, it has solved the issue. Many Thanks

MartyG-RealSense, 2021-08-01:
You are very welcome - that's great news.  Thanks for the update!

closed 2021-08-07 by MartyG-RealSense