realsenseai/librealsense #9535: D435i:  "no frames received" "depth stream start failure"

Lomasy, 2021-07-31:
I can not get Depth stream.
I bought a D435i, which can be used normally when I first bought it 2 weeks ago, but I can't use it now. I open the RGB camera in the realsense viewer and it can be used normally, but the stereo module can't read the depth picture, showing:“ no frames received”, and “depth stream start failure“.
I use it on another computer, but it still can't be used. I can see two cameras in the device manager: depth and RGB
[image]
[image]
[image]

MartyG-RealSense, 2021-07-31:
Hi @Lomasy  Thank you very much for the detail that you have provided that is useful for developing a diagnosis of your problem.

May I begin the investigation of your case by first asking the questions below, please.

1.  Have you updated the firmware driver of your D435i camera during the past two weeks?

2.  Have you had the camera placed unused in a place of storage during the past two weeks, such as in a desk drawer or a cupboard, and have only now started using the camera again and discovered this problem?

3. Does the problem still occur if you unplug the USB cable at its micro-sized end from the side of the camera, turn it around in the opposite direction and re-insert the connector into the side of the camera?  (USB-C cables are two-way insertion at the micro end).

Lomasy, 2021-08-01:
@MartyG-RealSense  
1.No.I didn't use the D435i in the past two weeks, so I didn't update the firmware. I used D435i yesterday, and I found that I couldn't get Depth, which is the problem I mentioned in the screenshot above.Then I updated the firmware to the latest one, but the problem remained unresolved. The screen after opening the viewer is the same as before upgrading.
2.Yes, I kept my camera in my locker and I didn't touch it, and only now started using the camera again and discovered this problem. The climate here is very dry. I don't think it is possible to damage the camera because of humidity. And I put the camera in the box provided by the manufacturer.And I can make sure that no one hits my locker or drops the camera on the ground with great force.
3.Yes, I did what you said, but the problem still exists, and the viewer still shows the same screenshot I provided above.

By the way, when I first got the camera two weeks ago, RGB stream was sometimes not received, then I would restart the computer to solve this problem. But most of the time it's good. So I thought RGB would be unstable.Now,  I didn't expect RGB to be very stable , but there was something wrong with depth.

MartyG-RealSense, 2021-08-01:
I asked about storage because there was a Windows case in June 2021 of depth stream start failure with No Frames Received for depth and infrared (RGB worked in that particular case).  The RealSense user had stored the camera unused in a cool dry area for two months, prior to which the camera had worked but it no longer worked when next used two months later.

I would go through a similar diagnostic procedure in your case, so it may be worth reading that case from the start and trying the range of tests that were suggested.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001316721-RealSense-D435-No-Frames-Received-in-Stereo-Module-Depth-Stream-

Lomasy, 2021-08-01:
@MartyG-RealSense 
Thanks a lot
I read the link you sent me carefully, and I tried all the steps. The results are as follows:
1. set the depth resolution in the real sense viewer to 480x270, and the result is the same as the original.
2. Reboot the computer and then launch the real sense viewer without opening any other programs. The result is the same as before
3. The result of using CustomRW tool to reset is the same as before
4.run rs-hello-realsense example program。 My error message is the same as Eniuyu's,
[image]

5. My windows has been updated, which is an update that comes with it. I also suspect that it has something to do with this, but I changed another computer with the same result
6.Hardware Reset, the result is the same as the original
7.check in the privacy settings 。 Privacy settings are fine.
8. check infrared stream。 I don't know how to  turn on IR stream
To sum up，
1. 5 and 8 may be right. if it is caused by windows update, is there any way to solve it? Or which windows version is more suitable for D435i? I can find a way to find the  computer to verify it
2. how to open the infrared stream in real sense viewer? I can't find the button
[image]

MartyG-RealSense, 2021-08-01:
You can test the infrared stream in the Viewer by expanding open the 'Stereo Module' options and left-clicking on the box beside **Infrared 1** to turn the box color from black (Off) to blue (On).  Then enable the streams with the icon beside 'Stereo Module'.

[image]

There is not an officially recommended version of Windows.  Because of the complexity of PCs, what works for one RealSense user might have problems on another's PC.  You can check your computer's update history in Windos Update.  You can find it by inputting **windows update** into the text box at the bottom of your Windows screen, next to the button in the corner where you shut down the PC.

On the Windows Update screen, select the **Check for updates** option to see recently installed updates.

[image]

Lomasy, 2021-08-02:
@MartyG-RealSense 
My IR stream doesn't work properly either.
[image]
So,My camera can not work.What should I do now?

MartyG-RealSense, 2021-08-02:
I will seek advice from Intel about your problem.  Thanks very much for your patience.

DLW3D, 2021-08-02:
Maybe it's the problem of default resolution, try to converts the default resolution of the depth camera to the same as the RGB camera.
https://blog.csdn.net/DLW__/article/details/119332367

MartyG-RealSense, 2021-08-08:
Hi @Lomasy  Does your camera still have a depth stream start failure, please?

closed 2021-08-25 by MartyG-RealSense