realsenseai/librealsense #9536: Human body scanning

websocketing, 2021-07-31:
Does your company's camera support size measurement after human body scanning?

Is there an SDK for measuring human body size for development？

MartyG-RealSense, 2021-07-31:
Hi @websocketing  If you require a 360 degree scan of the human body then it is certainly possible to arrange multiple RealSense cameras to capture all sides of the body.  Each camera can generate a 'point cloud' depth scan, and the individual point cloud scans can be 'stitched' together into a 360 degree single large point cloud.  

You can also measure the distance between two points.  Such multi-camera arrangements can be performed with **RealSense 400 Series** stereo depth cameras that can be used indoors and outdoors or with **RealSense L515** indoor lidar depth cameras.

*******************

If you would prefer to use a single camera and move it around the human subject then you can do so with a tool called **kinfu** that can build up a point cloud progressively as the camera is moved.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

Another single-camera measuring approach would be to point the camera at a specific area of the body that you wish to measure and use the pre-made **RealSense Viewer** tool's measuring option to drag out a line between two points on a point cloud and have a measurement of the distance automatically generated.

[image]

A similar point to point measuring tool can be created with program scripting.

******************

If you would prefer simple human height measurement from an RGB image instead of using depth data then RealSense can provide this through its compatibility with **TensorFlow**.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow

[image]

websocketing, 2021-08-01:
> 嗨@websocketing 如果您需要对人体进行360度扫描，那么当然可以安排几个摄像头来捕捉身体的各种自拍。 每天摄像头都生成“点云”深度扫描，并且可以将各个点云扫描“捕捉” ”到一个360度的重大点云中。
> 
> 您还可以测量两点之间的距离。多摄像头可以通过可在室内和室外使用的**实感****400系列**立体摄像头或**实感L515**室内激光雷达摄像头执行。
> 
> 如果您更喜欢使用特定的相机并在**身边**的对象周围移动它，那么您就可以使用**这些**工具来实现，该工具可以随着相机的移动而随时创建点云。
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu
> 
> 另一种单摄像头测量方法是将摄像头连接到您想要的身体特定区域，并使用预制的**RealSense Viewer**工具的测量选项在点云上的两点之间拖出一条线，并获得一条自动生成距离的测量值。
> 
> [image]
> 
> 可以使用程序脚本创建类似的点对点测量工具。
> 
> 如果您更喜欢对 RGB 图像进行简单的人体高度测量，那么 RealSense 就可以通过与**TensorFlow 的**一次来提供此功能。
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow
> 
> [image]

I intend to purchase dozens of depth cameras, but I want to know if your company's equipment has an SDK that can be used directly, because I don't want to write algorithms. I need to measure all parts of the human body.

websocketing, 2021-08-01:
@MartyG-RealSense  I intend to purchase dozens of depth cameras, but I want to know if your company's equipment has an SDK that can be used directly, because I don't want to write algorithms. I need to measure all parts of the human body.

MartyG-RealSense, 2021-08-01:
RealSense cameras do have a ready-made SDK called the **RealSense SDK 2.0**.  As mentioned above, the pre-made **RealSense Viewer** tool that is included for free with the RealSense SDK can be used to measure between two 3D points on a human in real-time.  The Viewer would not provide the ability to measure distances on all body areas simultaneously though, as it can only measure between two points at a time (though it can capture all of the body in the form of points).

 It should be technically possible though to retrieve anatomical measurements about multiple areas of the human body with RealSense, as the concept has been demonstrated with the commercial body scanning mirror product **Naked** (warning: image of a partially unclothed human in the link below).

https://www.intelrealsense.com/naked-home-body-scanner-based-on-intel-realsense/

websocketing, 2021-08-03:
@MartyG-RealSense What I want to know is, is there an SDK for d455 or d435 that can be called directly, or is it an algorithm?

MartyG-RealSense, 2021-08-03:
If the requirement was just for body scanning or capturing body motion then there would be a range of RealSense-compatible pre-made tools and SDKs that would meet that need.  If you need precise body part measuring though and need that process to be automated instead of selecting body areas manually, then that is likely to require programming your own application.  It should certainly be within the capabilities of RealSense camera technology and the RealSense SDK software though.

websocketing, 2021-08-05:
@MartyG-RealSense  Do you have any contacts or Chinese engineers for your products in China? I hope I can communicate with him.

MartyG-RealSense, 2021-08-05:
I will make enquiries for you about it.

websocketing, 2021-08-06:
@MartyG-RealSense  I need to determine whether your company's products and interface SDK can meet my needs, because at least I need a large number of hardware modules in the later stage. And I hope a Chinese can discuss with me, because I am used to asking in Chinese. My English is very poor.

MartyG-RealSense, 2021-08-06:
Please do write in Chinese language if that is comfortable for you.  I am pleased to respond in Chinese too.

如果你覺得舒服，請用中文寫。 我也很高興用中文回答。

websocketing, 2021-08-06:
@MartyG-RealSense 我想知道你们公司产品是否可以提供人体测量，例如身高，腰围，臀围等一系列的人体部位尺寸，自己写算法的话需要很长时间。或者你们可以提供有效的demo。

MartyG-RealSense, 2021-08-06:
Translation: "I want to know whether your company’s products can provide body measurements, such as height, waist circumference, hip circumference, and a series of body part sizes.  It will take a long time to write the algorithm by yourself.  Or you can provide a valid demo".

Hi @websocketing  In the PDF research paper linked to below, 13 RealSense D435 cameras were used for **anthropometric** measurements (measurement of the human body).

https://www.mdpi.com/1424-8220/20/13/3690/pdf

Page 14 of the paper has a table listing real body part perimeter measurements (R.1D) and their estimates (E.1D) in cm for a range of body areas.  The table is quoted in the image below.

[image]

An article on the official RealSense blog also highlights use of RealSense compatible products by the company **ImFusion** for body scanning and measurement applications.

https://www.intelrealsense.com/three-scanning-applications-with-imfusion/

**************

嗨@websocketing 在下面鏈接的 PDF 研究論文中，13 台 RealSense D435 相機用於**人體測量**測量（人體測量）。

https://www.mdpi.com/1424-8220/20/13/3690/pdf

該論文的第 14 頁有一個表格，列出了真實身體部位周長測量值 (R.1D) 及其對一系列身體區域的估計值 (E.1D)（以厘米為單位）。 該表在下圖中引用。

[image]

RealSense 官方博客上的一篇文章還重點介紹了 **ImFusion** 公司將 RealSense 兼容產品用於人體掃描和測量應用。

https://www.intelrealsense.com/three-scanning-applications-with-imfusion/

MartyG-RealSense, 2021-08-12:
Hello @websocketing  Do you require further assistance with this case, please?  Thanks!

你好@websocketing 請問這個案例你需要進一步的幫助嗎？ 謝謝！

MartyG-RealSense, 2021-08-18:
Case closed due to no further comments received.

由於未收到進一步評論，案件已結案。

closed 2021-08-18 by MartyG-RealSense