realsenseai/librealsense #9538: Depth map from back-projected point cloud is different from the original

zc08, 2021-07-31:
| Required Info | |
|---------------------------------|------------------------------------------- |
| Camera Model | D455 |
| Firmware Version | 05.12.06.00 |
| Operating System & Version | Ubuntu 18.04.3 |
| Kernel Version (Linux Only) | 5.4.0-48-generic |
| Platform | PC |
| SDK Version | 2.36.0 |
| Language | C/python |
| Segment | Robot |

### Issue Description
I'm using the following code to 1) align depth to color,  2) generate point cloud, 3) and recompute a depth map from the point cloud. 
I assume that the recompted depth map should be basically identical to the original depth map after alignment. However, the recompted depth is quite different from the original aligned depth. It seems there are some distortions present.

```
// align depth to color
rs2::depth_frame depth = frames.get_depth_frame();
rs2::video_frame color = frames.get_color_frame();
rs2::align align_to_color(RS2_STREAM_COLOR);
frames = align_to_color.process(frames);
depth = frames.get_depth_frame();
color = frames.get_color_frame();
// generate point cloud and points
rs2::pointcloud pc;
pc.map_to(color);
rs2::points pts = pc.calculate(depth);
// get depth map from points
cv::Size im_sz{int(depth.get_width()), int(depth.get_height())};
auto tex_coords = pts.get_texture_coordinates();
auto vertices = pts.get_vertices();
const int w = im_sz.width;
const int h = im_sz.height;
cv::Mat recomputed_depth = cv::Mat::zeros(h, w, CV_32FC1);
for (int i = 0; i < pts.size(); ++i) {
  rs2::texture_coordinate tex_coord = tex_coords[i];
  // convert normals [u v] to basic coords [x y]
  int x = std::min(std::max(int(tex_coord.u*w + .5f), 0), w - 1);
  int y = std::min(std::max(int(tex_coord.v*h + .5f), 0), h - 1);
  float *recomputed_depth_ptr = reinterpret_cast<float*>(recomputed_depth.data + y * recomputed_depth.step[0] + x * recomputed_depth.step[1]);
  *recomputed_depth_ptr = vertices[i].z;
}
// the recomputed depth is different from depth
```

This first image is the recomputed depth from the point cloud.
[image]

This second image is the depth frame after aligned to color.
[image]

It can be seen that for the first image, in the bright part on the upper-right side, some artifact curves present, while in the second image, there are none. And if you compare the two images by switching the windows back and forth, there is obvious distortion between them.

How doe this happen, what I'm doing wrong?

MartyG-RealSense, 2021-07-31:
Hi @zc08  One should bear in mind that when alignment takes place, the RealSense SDK's **Align processing block** changes the size of the field of view of the stream that is being aligned to match the FOV size of the align target.  So if depth is aligned to color then the depth FOV is resized to the color FOV.

zc08, 2021-07-31:
@MartyG-RealSense Yes I'm aware of that. I'm comparing the depth map *after* aligning to color stream, and the depth map obtained from point cloud (The point cloud is computed using the aligned depth map).

MartyG-RealSense, 2021-07-31:
I believe that the information in the llink below that is provided by a RealSense team member will be relevant to your application's scenario.

https://github.com/IntelRealSense/librealsense/issues/4644#issuecomment-522754917

closed 2021-08-13 by MartyG-RealSense