realsenseai/librealsense #9539: 640X480 for RGB of L515 is gone????

hwy1992129, 2021-07-31:
Is 640X480 for RGB option of L515 in Intel RealSense Viewer removed?
I was using 640X480 for RGB a few days ago.
[image]

MartyG-RealSense, 2021-07-31:
Hi @hwy1992129   Have you changed your L515's firmware version recently?  Support for 640x360 and 640x480 RGB resolutions was added in L515 firmware **1.5.8.1**

[image]

rampagedog12, 2021-08-01:
> Hi @hwy1992129 Have you changed your L515's firmware version recently? Support for 640x360 and 640x480 RGB resolutions was added in L515 firmware **1.5.8.1**
> 
> https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads
> 
> [image]

Thanks, I think I misclicked the install recommended firmware, and it rolled back to the old version................

closed 2021-08-07 by MartyG-RealSense