realsenseai/librealsense #9542: Camera not detected on Mac

Timmmm, 2021-08-01:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Don't have access currently but I updated it to the latest a couple of weeks ago. | 
| Operating System & Version |   MacOS  | 
| Platform                                 | Macbook Pro  |

### Issue Description

Hi, I have a D435 and I plugged it into my Macbook Pro via a Thunderbolt 3 dock (I don't have the right cable to go directly unfortunately). It is detected as a USB device in System Info, but no camera-using apps detect it (Zoom, Photo Booth, etc.). I'm trying to use it as a standard webcam.

I was under the impression that it presented itself as a standard UVC camera, and it definitely seems to behave like that on Windows - I was able to use it for Zoom on Windows without installing any drivers.

Would you expect it to work on Mac too?

MartyG-RealSense, 2021-08-01:
Hi @Timmmm  RealSense 400 Series cameras are UVC 1.5 compliant, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-470573083

Whilst I do not have access to a Mac to test this on, there are some video applications where the RGB feed of a RealSense camera is not visible unless a camera-change or view-change option is toggled.  An example of this is the built-in non-RealSense **Camera** app in Windows, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/8711#issuecomment-811945394

Zoom's Mac troubleshooting information states "If you are on on Mac OS 10.14 Mojave and are still having difficulty accessing the camera, check your operating system permissions to confirm that Zoom has access to the camera" (see the link below).

https://support.zoom.us/hc/en-us/articles/360016688031

Timmmm, 2021-08-02:
Hi, thanks for the reply!

> a camera-change or view-change option is toggled

Yeah I think that's just the normal camera selection UI right?

> check your operating system permissions

Unfortunately Mac's camera permissions are for all camera access (i.e. you grant access to all cameras, not just specific cameras) and I had already done that so that wasn't the issue.

I did some more testing - I found a USB-C to USB-C USB 3.1 cable so I could connect the camera directly to my Mac, but got the same result. I also tested on another Macbook which behaved exactly the same. It *is* detected, it just doesn't show up as a camera:

```
$ system_profiler SPUSBDataType
2021-08-02 09:46:53.285 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.286 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.286 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.287 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.287 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.288 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2021-08-02 09:46:53.288 system_profiler[63577:22599056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec
      PCI Revision ID: 0x0006
      PCI Vendor ID: 0x8086
      Bus Number: 0x00

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec
      PCI Revision ID: 0x0006
      PCI Vendor ID: 0x8086
      Bus Number: 0x01

        Intel(R) RealSense(TM) Depth Camera 435 :

          Product ID: 0x0b07
          Vendor ID: 0x8086  (Intel Corporation)
          Version: 50.ce
          Serial Number: 933623023949
          Speed: Up to 5 Gb/s
          Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
          Location ID: 0x01100000 / 1
          Current Available (mA): 900
          Current Required (mA): 720
          Extra Operating Current (mA): 0
```

(Those errors at the top happen without it plugged in at all so I assume it is unrelated.)

Unfortunately there seems to be almost no information available about debugging USB on Mac - nothing shows up in syslog. But I did eventually figure out that you can open the `Console` app (not the terminal), and click Start, and then I found these logs when I plug in the camera:

```
default	09:59:26.437452+0100	kernel	596693.199243 SSP1@01100000: AppleUSBHostPort::enumerateDeviceComplete_block_invoke: enumerated 0x8086/0b07/50ce (Intel(R) RealSense(TM) Depth Camera 435 )
default	09:59:26.509640+0100	VDCAssistant	[guid:0x110000080860b07] DeviceArrived service:0x6127 (id: 0x1000c6827)
error	09:59:26.511667+0100	VDCAssistant	CMIO_DPA_VDC_Server_Assistant.cpp:1453:GetDeviceByGUID No match for specified GUID 0x110000080860b07
error	09:59:26.511729+0100	VDCAssistant	DeviceArrived  caught an exception from determining the deviceType (0x2) or removing the device
default	09:59:26.511785+0100	VDCAssistant	[guid:0x110000080860b07] DeviceArrived  creating device (type: 2)
default	09:59:26.516032+0100	icdd	           Device DB | Creating local devices
default	09:59:26.517189+0100	VDCAssistant	[guid:0x110000080860b07] Found Camera with VID: 32902 (0x8086), PID: 2823 (0x0b07), release: 0x50ce
default	09:59:26.522507+0100	VDCAssistant	[guid:0x110000080860b07] Camera with bInterval 0 mConcurrentTransfers : 3
error	09:59:26.524117+0100	VDCAssistant	CMIO_DPA_VDC_Server_Device.cpp:556 cmio assert: "device != ((io_object_t) 0)" at exit - (err=0)
error	09:59:26.524281+0100	VDCAssistant	CMIO_DPA_VDC_Server_Device.cpp:2183:DiscoverCapabilities [guid:0x110000080860b07] DiscoverCapabilities: couldn't find any for format descriptors
default	09:59:26.518716+0100	kernel	Sandbox: icdd(623) allow file-read-data /Library/Image Capture/Devices
default	09:59:26.519176+0100	kernel	Sandbox: icdd(623) allow file-read-data /Library/Image Capture/Devices/EPSON Scanner.app/Contents/Resources/DeviceMatchingInfo.plist
error	09:59:26.524400+0100	VDCAssistant	CMIO_DPA_VDC_Server_Device.cpp:2224:DiscoverCapabilities IO return no device
error	09:59:26.524431+0100	VDCAssistant	DeviceArrived  Found a camera (0x110000080860b07) , but was not able to start it up (0xe00002c0 -- (iokit/common) no such device)
default	09:59:26.524533+0100	VDCAssistant	[guid:0x110000080860b07] DisposeFrames outstanding completions 0
default	09:59:26.524703+0100	VDCAssistant	[guid:0x110000080860b07] DisposeFrames outstanding completions 0
default	09:59:26.524913+0100	VDCAssistant	[guid:0x110000080860b07] DisposeFrames outstanding completions 0
default	09:59:26.524940+0100	VDCAssistant	[guid:0x110000080860b07] SuspendDeviceOnQuit false
```

Note the assertion. Could this be a bug in Mac's UVC driver, or the Realsense firmware? Can you guys confirm if this works for you or if it has ever been tested on Mac?

MartyG-RealSense, 2021-08-02:
The camera selector in video tools is designed for normal webcams but can work with RealSense cameras too.

A RealSense team member suggested in the link below that the librealsense SDK software would have to be installed on Mac in order for the camera to be accessed.

https://github.com/IntelRealSense/librealsense/issues/2893#issuecomment-447623904

Timmmm, 2021-08-02:
> The camera selector in video tools is designed for normal webcams but can work with RealSense cameras too.

Yes I am aware how camera selection UIs work 😃 

> A RealSense team member stated in the link below that the librealsense SDK software would have to be installed on Mac in order for a RealSense camera to be accessed as a general webcam.

No they're saying they were not able to access the camera via UVC at all, and the only way they could access it is via librealsense, which obviously doesn't work in Zoom, etc. because they don't use librealsense.

Installing librealsense can't make UVC work because it doesn't include any drivers. You can't "install" it at all in fact.

Thanks for linking to that thread though. It sounds like this is just broken and has probably never worked. It must be either a bug in Mac's UVC driver or a bug in the Realsense camera firmware.

It would be nice if you could fix it but I understand why it might be low priority!

MartyG-RealSense, 2021-08-02:
I researched the subject further but https://github.com/IntelRealSense/librealsense/issues/2893 seems to be the best reference available.  As I do not have a Mac to test with and do not have solid enough references to form confident advice on this particular situation, it will be better to see what advice Mac owners can provide in response to your comment on https://github.com/IntelRealSense/librealsense/issues/2893

In regard to librealsense not being able to make UVC work: the SDK can make use of the **RSUSB backend** on Mac, and RSUSB is compliant with UVC 1.5.  This is described in the SDK's **Build Configuration** documentation.

https://dev.intelrealsense.com/docs/build-configuration

MartyG-RealSense, 2021-08-08:
Hi @Timmmm  Do you require further assistance with this case, please?  Thanks!

Timmmm, 2021-08-08:
Well, I think it's probably unfixable by anyone except your firmware developers or Apple's UVC driver developers so I probably wouldn't spend any more time on it if I were you.

I did some more research though.

1. The libuvc/RSUSB situation in librealsense is pretty confusing! The `build-configuration` page you linked still says

> `FORCE_LIBUVC` - Setting this flag to true will configure the underlying backend to use libuvc. This backend is default for Mac OS, but can be configured on other OS-s as well. libuvc can serve as a more robust alternative to the native backend, however, it has well known limitations and is not recommended to be used in end-products (please use `FORCE_RSUSB_BACKEND` cmake flag instead)

But also

> `FORCE_RSUSB_BACKEND` - Equivalent to `FORCE_LIBUVC` ...

However looking at the code it seems like a couple of years ago libuvc was removed entirely in favour of RSUSB (which I guess is your own version?

2. I actually tried using upstream libuvc to open the camera but it fails due to some UVC 1.5 issues, so I'm not sure how librealsense ever used libuvc to do it, unless you patched it? I opened a bug in libuvc about that [here](https://github.com/libuvc/libuvc/issues/189).

MartyG-RealSense, 2021-08-08:
As far as I am aware, the **FORCE_LIBUVC** instructions are deprecated rather than removed completely, meaning that they are usable but it is recommended to use the RSUSB instruction instead.  An example of installing via **libuvc-backend** (which dates back to before the introduction of RSUSB in librealsense SDK version 2.30.0) can be found in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

In Linux, a non-RSUSB build (**-DFORCE_LIBUVC=OFF**) uses **V4L2 backend** instead of RSUSB backend.

https://github.com/IntelRealSense/librealsense/issues/6841#issuecomment-660859774

Timmmm, 2021-08-08:
Those instructions and script are out of date and should have been removed when the libuvc backend was removed. If you check the code there is no way to use libuvc now and `FORCE_LIBUVC` is just an alias to enable RSUSB.

I guess it's possible that some old firmware used to work with libuvc, but new firmware doesn't and nobody noticed because librealsense can't use libuvc anymore.

Anyway kind of getting off topic!

Timmmm, 2021-08-14:
Nope. I mean it's still broken but I don't need it fixed.

MartyG-RealSense, 2021-08-14:
Okay, thanks very much @Timmmm for the update.  I will close the case, but please feel free to return to it at a future date if you wish to do so.  Good luck!

closed 2021-08-14 by MartyG-RealSense