realsenseai/librealsense #9543: When I use m_device.setImageRegistrationMode(openni::IMAGE_REGISTRATION_DEPTH_TO_COLOR) in openni to align the depth image and the color image, the frame rate will drop severely.

zma69650, 2021-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455 }  | 
| Firmware Version                   | 05.12.13.50  | 
| Operating System & Version |   {Win10   | 
| Kernel Version (Linux Only)    |  无                           | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<45>.<0> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When I use m_device.setImageRegistrationMode(openni::IMAGE_REGISTRATION_DEPTH_TO_COLOR) in openni to align the depth image and the color image, the frame rate will drop severely.

MartyG-RealSense, 2021-08-02:
Hi @zma69650  Alignment of color and depth is a processing-intensive activity that may result in slowdown on some computers.

Are you using the **SimpleVewer** OpenNI example program to perform registration or incorporating IMAGE_REGISTRATION_DEPTH_TO_COLOR into your own script, please?

MartyG-RealSense, 2021-08-02:
Could you provide a link to the OpenNI sample alignment program that you are referring to please to make sure that we are both talking about the same program?  Thanks very much.

zma69650, 2021-08-03:
This is an example provided by openni. https://github.com/OpenNI/OpenNI2/blob/master/Samples/SimpleViewer 
There is an alignment option in the OpenGL keyboard response function.
'''

	case '1':
		m_eViewState = DISPLAY_MODE_OVERLAY;
		m_device.setImageRegistrationMode(openni::IMAGE_REGISTRATION_DEPTH_TO_COLOR);
		break;

'''

MartyG-RealSense, 2021-08-03:
So the alignment part is optional because it is activated by a keypress and you could strip that part out of the program if you do not require alignment (as https://www.stereolabs.com/docs/openni2/#linux-examples highlighted that depth-only and color-only streaming can be performed in OpenNI)

closed 2021-08-06 by zma69650