realsenseai/librealsense #9544: D435i depth distance accuracy

b2992872, 2021-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            |  { 2.16.1 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
In the last question #9447 , my environment was built, and then I executed the following example, and found that the “depth_frame.get_data” method used to obtain the depth, but there is a big error with the actual distance, which may reach tens of centimeters, and I It has been calibrated through imu.
`import cv2
import re
import numpy as np
import jetson.inference
import jetson.utils
import argparse
import sys
import os
import io
import time
import json
import random
import pyrealsense2 as rs

# parse the command line

parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage() +jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("--network", type=str, default="ssd-mobilenet-v2",help="pre-trained model to load (see below for options)")

parser.add_argument("--threshold", type=float, default=0.5,help="minimum detection threshold to use")

parser.add_argument("--width", type=int, default=640,help="set width for image")

parser.add_argument("--height", type=int, default=480,help="set height for image")

opt = parser.parse_known_args()[0]
# load the object detection network

net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# Configure depth and color streams

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, opt.width, opt.height, rs.format.z16, 30)
config.enable_stream(rs.stream.color, opt.width, opt.height, rs.format.bgr8, 30)

# Start streaming

pipeline.start(config)

os.system("sudo sh -c 'echo 128 > /sys/devices/pwm-fan/target_pwm'")

press_key = 0
while (press_key==0):
	
   # Wait for a coherent pair of frames: depth and color

   frames = pipeline.wait_for_frames()
   depth_frame = frames.get_depth_frame()
   color_frame = frames.get_color_frame()
   if not depth_frame or not color_frame:
       continue	
   # Convert images to numpy arrays

   depth_image = np.asanyarray(depth_frame.get_data())
   color_image = np.asanyarray(color_frame.get_data())
   depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
   show_img = np.asanyarray(color_frame.get_data())
   
   gray = cv2.cvtColor(color_image, cv2.COLOR_BGR2GRAY)

   # convert to CUDA (cv2 images are numpy arrays, in BGR format)
   bgr_img = jetson.utils.cudaFromNumpy(show_img, isBGR=True)

	
   # convert from BGR -> RGB
   img = jetson.utils.cudaAllocMapped(width=bgr_img.width,height=bgr_img.height,format='rgb8')

   jetson.utils.cudaConvertColor(bgr_img, img)

   # detect objects in the image (with overlay)

   detections = net.Detect(img)
   for num in range(len(detections)) :
       score = round(detections[num].Confidence,2)
       box_top=int(detections[num].Top)
       box_left=int(detections[num].Left)
       box_bottom=int(detections[num].Bottom)
       box_right=int(detections[num].Right)
       box_center = detections[num].Center
       label_name = net.GetClassDesc(detections[num].ClassID)

   point_distance=0.0
   for i in range (10):
       
       point_distance =  depth_frame.get_distance(int(box_center[0]),int(box_center[1]))

       point_distance = np.round(point_distance , 3)

       distance_text = str(point_distance) + 'm  ' + str(score)+ '%'

       cv2.rectangle(show_img,(box_left,box_top),(box_right,box_bottom),(255,0,0),2)

       cv2.line(show_img,(int(box_center[0])-10, int(box_center[1])),(int(box_center[0]+10), int(box_center[1])),(0, 0, 255), 3)

       cv2.line(show_img,(int(box_center[0]), int(box_center[1]-10)),(int(box_center[0]), int(box_center[1]+10)),(0, 0, 255), 3)

       cv2.putText(show_img,label_name + ' ' + distance_text,(box_left+5,box_top+20),cv2.FONT_HERSHEY_SIMPLEX,0.4,(0,0,255),1,cv2.LINE_AA)

       cv2.putText(show_img,"{:.0f} FPS".format(net.GetNetworkFPS()),(int(opt.width*0.8), int(opt.height*0.1)),cv2.FONT_HERSHEY_SIMPLEX,1,(0,0,255),2,cv2.LINE_AA)

       display = cv2.resize(show_img,(int(opt.width*1.5),int(opt.height*1.5)))

       images = np.hstack((color_image, depth_colormap))
       #cv2.imshow('Detecting...',display)
       cv2.imshow('RealSense', images)

   keyValue=cv2.waitKey(1)
   if keyValue & 0xFF == ord('q'):
       press_key=1

# 

cv2.destroyAllWindows()
pipeline.stop()
`
[image]

MartyG-RealSense, 2021-08-02:
Hi @b2992872   RealSense 400 Series cameras have separate calibration tools for depth / color and for IMU.  The IMU calibration tool would not calibrate depth and color.  

For depth and color calibration, you can use the RealSense SDK's **Dynamic Calibration** tool.  It can be installed on Linux using the instructions in page 14 onwards of the tool's PDF user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

RealSense 400 Series cameras are calibrated in the factory during the manufacturing process and so have very good calibration already when new out of the box, though may need recalibration if they receive a physical shock such as a hard knock or a drop on the floor.

Having said that though, in this particular case your incorrect measurements may be due to the number of black objects in the scene.  It is a general physics principle (not specific to RealSense) that dark grey and black surfaces absorb light and make it harder for a depth camera to analyze those surfaces for depth detail.  The darker the shade, the less information that is returned.   

If it is unavoidable to have black objects in the scene, projecting a strong light source such as a lamp onto the objects will help to bring out depth detail, as demonstrated in the case with a black office chair in the link below.

https://github.com/IntelRealSense/librealsense/issues/6757

b2992872, 2021-08-03:
In order to avoid color problems, I changed to recognize people, but when the person is 2 meters away from the lens, the distance is not visible, and when the person is 1 meter away from the lens, the distance is 3.25 meters, because the depth image features are not significant, and the recognition distance will be Is it not accurate?
[image]
[image]

MartyG-RealSense, 2021-08-03:
I believe that the DNN detection of a person is working okay but it is still failing on accurate depth detection because of the dark grey shirt color under the box-center cross (remember, the camera has difficulty with reading black **and** dark grey).  There was once a case where a person was wearing a shirt with red and grey stripes, and only the red stripes registered as depth detail and the grey stripes were rendered as areas of empty depth.

b2992872, 2021-08-04:
I used another D435i test on hand and found that the error was about 1cm. I think the original D435i needs to be dynamically corrected.

MartyG-RealSense, 2021-08-04:
Thanks very much @b2992872 for the update.  Please do let me know if the original D435i has its results improved after a dynamic calibration.  Good luck!

closed 2021-08-08 by b2992872