realsenseai/librealsense #9545: Unable to build Unity wrapper

lafith, 2021-08-02:
I am trying to build the Unity wrapper based on the instructions given [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#step-1---build-dependencies-and-open-the-unity-project). Unfortunately running cmake is giving an error as shown in the image.

[image]

What should be the correct command?

MartyG-RealSense, 2021-08-02:
Hi @lafith  There is an easier method to install the RealSense Unity wrapper.

1.  Download the file **Intel.RealSense.unitypackage** from the 'Assets' file list at the bottom of the RealSense SDK version listing at the SDK **Releases** web-page.

https://github.com/IntelRealSense/librealsense/releases

[image]

2.  Create a new Unity project or load your existing one.  Then go to the **Assets > Import Package > Custom Package** menu option and use the file selector to select the Intel.RealSense.unitypackage file that you downloaded to your computer.  It will typically be located in the **Downloads** folder of your Windows user documents folder if you did not define a custom path when downloading it.

Another way to find the Downloads folder is to left-click on the arrow beside the file download tab at the bottom of your browser screen and select the **Open file location** menu option.

[image]

3.  After you select the Intel.RealSense.unitypackage file, Unity will begin automatically import the RealSense Unity wrapper files into your project.  When a pop-up list of files appears, left-click on the **Import** button at the bottom tof the panel to complete setup of the wrapper in your Unity project.

4.  To launch the starter project, go to the central-bottom **Assets** panel of the Unity window and naviate to **Assets > RealSenseSDK2.0 > Scenes** and double-leftclick on the **StartHere** scene file to load the starter project.  Once you have the RealSense camera plugged in, left-click on the 'play' triangle at the top of the Unity screen to run the project.

[image]

lafith, 2021-08-02:
Thank you, that worked for importing. Following is the output on selecting 'UI Image Sample(Depth & Color).
[image]
[image]

MartyG-RealSense, 2021-08-02:
Hi @lafith  Which RealSense camera model do you have plugged in when you run the example, please?  Also, which RealSense SDK version do you have installed?

lafith, 2021-08-02:
Camera Model: D435
SDK version : Intel® RealSense™ SDK 2.0 (v2.47.0)

MartyG-RealSense, 2021-08-02:
Thanks for the information.  That setup should be fine.  I just wanted to confirm that you were using a RealSense SDK version where Unity 2020 was supported, and you are.

Have you got any other programs using the camera at the time that you run the Unity sample, such as the RealSense Viewer?  RealSense programs currently streaming with the camera should be closed before running another RealSense program, because the program that was run first will be blocking access to other applications using the streams that it currently has a 'claim' on.

lafith, 2021-08-02:
Yes, that fixed it.

closed 2021-08-02 by lafith