realsenseai/librealsense #9548: Location of realsense-viewer

rolandgvc, 2021-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.44 }                          | 

### Issue Description
Where is the realsense viewer executable located at? After removing the SDK according to
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages
the realsense viewer can still be called.

MartyG-RealSense, 2021-08-02:
Hi @rolandgvc  Please look in the **usr/local/bin** folder for the built examples and tools.

rolandgvc, 2021-08-02:
Excelent, thank you

closed 2021-08-02 by rolandgvc